nickcoutsos/keymap-editor #5: Document zmk binding behaviour definitions

nickcoutsos, 2021-10-05:
Some of the current `zmk-behaviours.json` is pretty straightforward and don't need much explanation. Other parts (`"commands"` and, more specifically, `"additionalParams"`) are my own invention and while they get the job done for now, they could do with some refinement and justification.

jackytrabbit, 2023-03-12:
After tried to use tap hold/dance features in the beta version, I think descriptions for parameters on behaviour settings. Some folks like me turning to ZMK were using QMK/VIA/Vial before. The ZMK behaviour part is a little complicated for them. Of course Keymap-Editor is not a standalone tool. We should check ZMK doc for more info in order to use those behaviours. But if there are descriptions for parameters, that would be nice.

nickcoutsos, 2023-03-12:
Hi, thank you for the feedback!

Firstly, I really appreciate your understanding that the editor isn't a substitute for understanding ZMK. I do understand how confusing ZMK can be (it's really only through building this that I've started to grasp some of these details), and I think that's partly due to its flexibility.

In order to match that flexibility in the app I need to keep things as generic as I can -- at least to start. There is room for having behavior-specific editors where a clearer/smoother experience is possible, but that takes effort so it comes down to priorities. I think adding brief parameter descriptions could be helpful, though I would want to avoid trying to duplicate ZMK documentation and getting into a position where I need to maintain another version of that knowledge. A simpler starting point might even mean linking to specific behavior documentation pages.

nickcoutsos, 2023-03-12:
I forgot how old this issue was -- it's actually intended for documentation intended for developers. I'd like to know a little more about the difficulties you've encountered and what specifically you had in mind, would you want to create a new issue to continue this discussion?

jackytrabbit, 2023-03-13:
> I forgot how old this issue was -- it's actually intended for documentation intended for developers. I'd like to know a little more about the difficulties you've encountered and what specifically you had in mind, would you want to create a new issue to continue this discussion?

Thanks for your reply! I totally understand your consideration on those points.
I will create a new issue on trying new beta features.

closed 2023-08-02 by nickcoutsos