nickcoutsos/keymap-editor #19: Configuration changes

ReFil, 2022-01-04:
Config variables are in board_defconfig or shield.conf depending on if it's a board or a shield, this could be moved to the top of config folder to be used and manipulated by the editor. config variables to be changed could be defined in info.json to be edited with bool ones changed using switches and int ones changed with a simple number box

nickcoutsos, 2022-01-04:
Thanks for surfacing this here, I haven't had a need to touch these files yet personally so I'm unaware of their intricacies so far. You mention a couple of files though I'm seeing four in my repo (not counting the split left/right files). In my case it's a board/shield though I couldn't even say if all of those files even do anything or are necessarily recognized by ZMK in their current path. All of this is to explain my hesitation, but maybe these questions would be more helpful for now:

1. What syntaxes are used in these files? Would they need to be parsed to preserve arbitrary existing configurations or would the expected behaviour involve the editor overwriting files on commit? It sounds like you intend for the latter but I'm seeing more and more friction with trying to make custom JSON schemas for the subset of things that currently make sense in order to spit out actually code during the build (for example, behaviour definitions in the keymap).
2. Are these config options exhaustively described somewhere? I see mentions of them spread around the ZMK docs but I imagine these are more like typical examples and not necessarily everything someone might want to control (which again raises the question of how to handle configuration the the editor doesn't yet understand without clobbering the user's existing config).

ReFil, 2022-01-05:
All config options are explained in the root kconfig file in zmk, but I don't think being able to arbitrarily change all of them is necessary. A lot of them are required for the board to function like enabling gpio and so on. I'm viewing it from an integrator/manufacturers standpoint whereby they can define what configuration options make sense for the end user to change and what should be left alone, but they could all be parsed and made available for editing

ReFil, 2022-01-05:
https://github.com/PolarityWorks/editor-zmk-config/blob/bt65-ANSI/config/boards/arm/bt65/bt65_defconfig

Here's an example defconfig, as you can see a lot of the stuff doesn't need to be played with but things like RGB default settings and backlight power management options definitely can be

nickcoutsos, 2022-04-12:
I'm going to go ahead and close this but feel free to reopen if you want to continue the discussion.