nickcoutsos/keymap-editor #21: Export/Print as PDF

kpalatzky, 2022-03-25:
It would be very helpful to be able to export all layers simultaneously in one PDF.

nickcoutsos, 2022-03-27:
Yeah I can see the benefit, I've considered adding it in the past. PDF generation can be a pain, but maybe PNG?

Either way working on this project alone has been a slog and I haven't been able to motivate myself to work on it as much lately, so I can't tell you how soon a feature like this might be available.

nickcoutsos, 2022-05-27:
Hi @kpalatzky 

If you're still around I've been doing a bunch of work on the keymap editor recently and getting to the point where I can see myself working on this feature in the not too distant future.

Again, I probably wouldn't go with a PDF but a PNG could be useful. Especially for embedding in the repository's `README.md`. I wouldn't want the app making changes to the readme but as long as the image(s) location is predictable then the repo owner can use it however they want.

Dimentium, 2022-07-08:
hello!
just as a suggestion:
is it possible to add "view mode" in addition to "editor mode", or maybe "show all layers mode"?
so that you can just make a screenshot of the page with all layers or "print as pdf" from the browser.

kpalatzky, 2022-07-08:
My idea here would have been to work with PrintCSS. With this you could probably take the current HTML, join & tansform it a bit and use the current CSS to create a PDF.

nickcoutsos, 2022-07-09:
> is it possible to add "view mode" in addition to "editor mode", or maybe "show all layers mode"?

I don't know that I'd make it its own "mode" exactly, but I've been looking into re-doing the keymap component to match the combo and macro editors. So instead of managing a single layer at a time, have one big scrollable list of each layer with the same edit/delete buttons. If I get it to that point I suppose it might be possible for a "hide UI" feature of some kind.

> My idea here would have been to work with PrintCSS. With this you could probably take the current HTML, join & tansform it a bit and use the current CSS to create a PDF.

Hmm. I haven't seen that before. Admittedly it would likely get the job done, but I'd have to do some more research.

nickcoutsos, 2023-08-28:
I'm starting to think that a full blown diagram export or standalone renderer will most likely not happen, see: https://github.com/nickcoutsos/keymap-editor/issues/114

I'd still like to get to the point where rendering layers sequentially in the app makes sense and that could lend itself to a printable view, but I have a lot of other things to figure out first.

closed 2023-08-28 by nickcoutsos

nickcoutsos, 2023-09-27:
I've started a [discussion](https://github.com/nickcoutsos/keymap-editor/discussions/141) on this topic to get more input in one place.