nickcoutsos/keymap-editor #23: Please add Label Name and Sensor Bindings in .keymap

LamDVT, 2022-04-05:
We need `label = "LayerName";` for oled display and `sensor-bindings = <&inc_dec_kp C_VOL_UP C_VOL_DN>;` for Rotary Encode
Also need add dynamic for sensor-bindings
Just replace line 100 to 107 in  /api/services/zmk/keymap.js
```
return `
        ${name.replace(/[^a-zA-Z0-9_]/g, '_')} {
			label = "${name.replace('layer', '')}";
            bindings = <
${rendered}
            >;
			sensor-bindings = <&inc_dec_kp C_VOL_UP C_VOL_DN>;
        };
`
  })
```

nickcoutsos, 2022-04-05:
The label seems reasonable enough, but I don't like hardcoding sensor-bindings as a solution for proper support for rotary encoders.

> We need `label = "LayerName";` for oled display and `sensor-bindings = <&inc_dec_kp C_VOL_UP C_VOL_DN>;` for Rotary Encode

Who is "we"?

LamDVT, 2022-04-05:
I know that hardcoding is not acceptable. That was reason I wrote `Also need add dynamic for sensor-bindings`
`Who is "we"?` my mistake, just me.

nickcoutsos, 2022-04-05:
Sorry, I focused on the code snippet and jumped to the wrong conclusion as it does really trivialize the problem. There's a lot of unanswered questions to make this work dynamically (also, how is this presented in the UI?) that I've been documenting in a previous issue https://github.com/nickcoutsos/keymap-editor/issues/12 if you want to help with some ideas or a pull request.

nickcoutsos, 2022-04-11:
Hi @LamDVT 

There was some work in progress in editing the devicetree code directly rather than overwriting it with a pre-set template and while I'm not close to proper support for rotary encoders I have a feature that may help you for now.

I have just pushed some experimental support for parsing your existing `*.keymap` file and replacing _just_ the layer bindings. This means if you prepare a keymap that includes `sensor-bindings` and `label` properties in your layers then you can still use the keymap editor to tweak key bindings without losing the rest of the keymap content.

If you want to try it out make sure to toggle the `Enable Devicetree Parsing` setting
[image]
_(hit the gear icon in the top-right to open the settings dialog)_

And, of course, this is still pretty rough. I've only had enough time for some quick and dirty testing so far. If you run into any problems please feel free to raise an issue (separately from this one, preferably) and I'll see what I can do to help!

nickcoutsos, 2022-06-28:
Hi @LamDVT, if you're still around and using the editor I've added support for editing sensor bindings. I don't have a keyboard of my own with rotary encoders built in so if you end up trying this feature out I'd be interested in knowing how you find the new functionality.