nickcoutsos/keymap-editor #26: Support dynamic keymap selection

nickcoutsos, 2022-04-12:
Assuming that in a given repository or `ZMK_CONFIG` workspace any editable .keymap file will found under `config/` regardless of board/shield stuff, we should be able to dynamically fetch this information and present it as another selection just like the GitHub picker does with branches.

Doing this once means the keymap name becomes an identifier of sorts so we don't have to do that same "search" when committing changes to the repo.

Using the matrix in `build.yaml` _might_ be useful but this is unclear to me still.

ankushg, 2022-09-28:
Seconding this feature request!

For a more detailed look at how I was thinking of doing things:

## Current Setup

### Directory structure

I currently have

```
config/
├─ adv360.keymap           # contains the single keymap for the branch
├─ adv360_left.keymap      # contains `#include "adv360.keymap"`
├─ adv360_right.keymap     # contains `#include "adv360.keymap"`
```

### Usage:
- Use the GUI to edit `adv360.keymap`
  - conveniently this gets automatically selected -- not sure if I'm just getting lucky because of alphabetical ordering
- create and maintain a separate branch for any different keymaps
  - becomes hard to share macros and other customizations

## Desired Setup
```
config/
├─ active.keymap           # modified to contain `#include "config/keymap/<whatever>.keymap"`
├─ adv360_left.keymap      # contains `#include "active.keymap"`
├─ adv360_right.keymap     # contains `#include "active.keymap"`
├─ keymaps/                # contains multiple keymaps within a single branch
│  ├─ adv360.keymap
│  ├─ ergodox-ish.keymap
│  ├─ ordinary.keymap

```

Usage:

- use the GUI to explicitly choose which `config/keymap/*.keymap` file to edit
- maintain multiple keymaps within a single branch
- declare a desired keymap before building by defining `active.keymap` to `#include "keymaps/whatever.keymap"` (or copy the desired keymap file to overwrite `active.keymap`)
	- This is also be where the build matrix can come in handy: 
		- iterate over every file in `config/keymaps/*`
		- copy to `active.keymap`
		- build!

nickcoutsos, 2022-09-28:
> I currently have [...]

I believe the intended structure is

```
config/
├─ boards/    <-- this ideally lives in an upstream ZMK repo/fork
│  ├─ adv360/
│  │  ├─ adv360.keymap
│  │  ├─ adv360_left.keymap
│  │  ├─ adv360_right.keymap
├─ adv360.keymap    <-- the build process merges this onto the "default" keymap devicetree
```

> conveniently this gets automatically selected -- not sure if I'm just getting lucky because of alphabetical ordering

You are, but again this isn't the preferred way to handle split keyboards.

The idea of selectable keymaps here is to support users with multiple keyboards in a single `zmk-config` repository, not to have one keyboard with alternate keymaps for different use cases. I think the build config can get you the same result but it's not something I'm thinking about.

nickcoutsos, 2023-03-04:
Forgot about this -- I added support for zmk-config repos with multiple keymaps around a month back!

closed 2023-03-04 by nickcoutsos