nickcoutsos/keymap-editor #27: Command failed (Running Locally)

or-agami, 2022-04-13:
Hi,

I tried to install the app with [this guide[](https://github.com/nickcoutsos/keymap-editor/blob/d61a71b2882e203d3c2dde9532e404f54b122986/running-locally.md) and I faced with Error "Command failed: npm run build-watch", any advice? 

[image]

nickcoutsos, 2022-04-13:
That's... that's new to me. Looking into the `ERR_OSSL_EVP_UNSUPPORTED` pulls up some issues:

* https://www.bswen.com/2021/11/reactjs-ERR_OSSL_EVP_UNSUPPORTED_error_solution.html
* https://stackoverflow.com/questions/69394632/webpack-build-failing-with-err-ossl-evp-unsupported

I haven't seen this come up, I'm on Node v17.8.0, but would you want to try `export NODE_OPTIONS=--openssl-legacy-provider` suggestion and let me know how that goes? If it doesn't pass into the build-watch process you may need to add it to the env mapping in `api/routes/application.js` line ~16.

nickcoutsos, 2022-04-13:
What's weird is that I keep seeing mention of this issue relating to webpack@4 but I've got version 5 in the dependencies. I see webpack-cli is a major version of 4, wonder if that's got something to do with this.

nickcoutsos, 2022-04-21:
@or-agami hi, was there any luck with the above links?

or-agami, 2022-04-22:
@nickcoutsos Actually I didn't tried again I've been busy at work.
I'll update as soon as I have something, thanks!

or-agami, 2022-04-22:
I think I messed up with the SSL files somehow

nickcoutsos, 2022-04-22:
Oh that's an interesting problem. Not sure what to do about that.

closed 2022-05-20 by nickcoutsos