nickcoutsos/keymap-editor #29: Local install issue

linkdevk, 2022-04-23:
Hello, this project looks awesome. 

I am having an issue installing locally. I cloned the repo, npm install, then npm run dev and I get the error  

`Z:\web dev\github\keymap-editor>npm run dev

> keymap-editor@1.0.0 dev
> ENABLE_DEV_SERVER=true node index.js

'ENABLE_DEV_SERVER' is not recognized as an internal or external command`

Any pointers? Thanks!

nickcoutsos, 2022-04-23:
That's an environment variable. They work differently on windows, unfortunately, I've seen people use `cross-env` to work around that I should probably add that myself at some point. In the meantime, if you can set the environment variable yourself you can just do the windows equivalent of that script.

```
set ENABLE_DEV_SERVER=true
node index.js
```

Something like that?

nickcoutsos, 2022-04-23:
Just pushed up a commit if you want to try that. I don't have a dev environment on my Windows machine so let me know how it goes!

linkdevk, 2022-04-23:
Hey thanks, I just tried your commit. After npm install and then npm run dev I get this error

`node:internal/fs/utils:344
    throw err;
    ^

Error: ENOENT: no such file or directory, open 'Z:\web dev\github\keymap-editor\private-key.pem'
    at Object.openSync (node:fs:585:3)
    at Object.readFileSync (node:fs:453:35)
    at Object.<anonymous> (Z:\web dev\github\keymap-editor\api\services\github\auth.js:8:61)
    at Module._compile (node:internal/modules/cjs/loader:1101:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1153:10)
    at Module.load (node:internal/modules/cjs/loader:981:32)
    at Function.Module._load (node:internal/modules/cjs/loader:822:12)
    at Module.require (node:internal/modules/cjs/loader:1005:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (Z:\web dev\github\keymap-editor\api\services\github\index.js:8:5) {
  errno: -4058,
  syscall: 'open',
  code: 'ENOENT',
  path: 'Z:\\web dev\\github\\keymap-editor\\private-key.pem'
}`

nickcoutsos, 2022-04-23:
Hey thanks for pointing this out. It's been a while since I've had to start from scratch and I've had very little feedback from others using the code :)

There's some required parameters for the GitHub integration but it sounds like maybe you aren't looking for that so I've pushed some changes to conditionally import those routes and did some general configuration cleanup. I tried a fresh clone of the repository with the change and was able to get things running... some of the documentation may be a little out of date but hopefully all the important stuff is there to guide you.

linkdevk, 2022-04-24:
I was able to get it running with wsl. Doesn't want to run on windows. I appreciate your help! The app is awesome thanks for putting this up 👍 

If you're interested here is the error I got when running it on straight windows. 

```Z:\web dev\github\keymap-editor>npm run dev

> keymap-editor@1.0.0 dev
> cross-env ENABLE_DEV_SERVER=true node index.js

listening on 8080
Error: spawn npm ENOENT
    at Process.ChildProcess._handle.onexit (node:internal/child_process:282:19)
    at onErrorNT (node:internal/child_process:477:16)
    at processTicksAndRejections (node:internal/process/task_queues:83:21) {
  errno: -4058,
  code: 'ENOENT',
  syscall: 'spawn npm',
  path: 'npm',
  spawnargs: [ 'run', 'build-watch' ],
  cmd: 'npm run build-watch'
}
Application serving failed```

nickcoutsos, 2022-04-24:
Oh that's really bothersome. I found [this](https://stackoverflow.com/questions/43230346/error-spawn-npm-enoent) related question. Would you be at all interested in trying a fix?

On line 24 of `api/route/application.js` try changing `'npm'` to `'npm.cmd'`. I vaguely remember Windows being really obnoxious about which npm binary was available in different contexts.

linkdevk, 2022-04-24:
That worked!

nickcoutsos, 2022-04-24:
Awesome! I'll make another issue to track selecting `npm`/`npm.cmd` as appropriate though I may not get to it that quickly.

If you don't mind sharing, what is your zmk-config setup like? I'm doing more work these days on the GitHub integration so the main branch and the local source may be a little behind in features. It is possible to use the hosted app with a private GitHub repo, though understandably there are still some situations it doesn't support fully. No pressure!

linkdevk, 2022-04-26:
What would you like to know about my setup? 

I am using a Sofle V2 board.

Where does the info.json file come from? Is that from QMK? Is it possible to create a keymap in your app from my existing .keymap file?

nickcoutsos, 2022-04-26:
Just wondering if there are features you need from a local setup specifically.

The `info.json` is an extension of QMK's, adding `col` and `row` properties that I use to align the generated keymap text. The other details are used to lay the keys out in the web UI.

I'm working on experimental support for repositories that don't include an `info.json` file. This involves looking for a board definition `config/boards/shields/<keyboard>/<keyboard>.(dtsi|overlay)` or `config/boards/arm/<keyboard>/<keyboard>.dts`), extracting the matrix-transform node and parsing the defined map into a layout. The code is not public yet but it's usable on the hosted editor (using the GitHub integration) and works pretty well. Better for layouts that are already represented ortholinearly in ZMK, but the point is to just get things started.

The drawback to this is the requirement that this ZMK keyboard definition exists in the repository. A lot of times well known keyboards will have these core files in the zmkfirmware/zmk repo and the build script in a user's zmk-config repo will fetch that. This seems to be the case for Sofle, and at this time the keymap editor doesn't do that.

linkdevk, 2022-04-26:
How do you create an info.json file? Will the Github integration generate the layout for Sofle out of the box without an info.json?

nickcoutsos, 2022-04-26:
If you want to create one it's [described in the QMK docs](https://docs.qmk.fm/#/reference_info_json). Short of that, you can look at the QMK repo and see if there's one created already for the Sofle (if you can edit the keymap with the QMK Configurator then this file should exist).

This keymap editor will look for `config/boards/shields/sofle/sofle.dtsi` in your repository (assuming your keymap file is `config/sofle.keymap`). If it can't find that file then it won't attempt to generate the layout.

closed 2022-05-08 by nickcoutsos