nickcoutsos/keymap-editor #38: Support for mod-morph

rambip, 2022-07-10:
Is is currently possible to create [mod-morph](https://zmk.dev/docs/behaviors/mod-morph) behaviours in the web-app ?

My use-case would be to change `SHIFT+COMMA` to output `SEMICOLON`

Is there a way to do it ?
If not, will that be supported in the future ?

nickcoutsos, 2022-07-10:
Hi @rambip 

At this time I don't have support for making any changes to behaviour customizations in-app. I would like to, I think, but I haven't made concrete plans about how to tackle it. Different behaviours have different levels of complexity, some just have a few properties to edit, but I want to make sure I understand the ones that require a list of `bindings`. The ZMK source describes these behaviours a little in yaml schemas ([zmk,behavior-mod-morph.yaml](https://github.com/zmkfirmware/zmk/blob/main/app/dts/bindings/behaviors/zmk%2Cbehavior-mod-morph.yaml) for example) so I'll investigate that a bit more to start.

This may not be what you're looking for but, for now, I'm hoping that these kinds of customizations don't need to happen too frequently in your keymap. If that's the case, you can still manually create the behaviour and the keymap editor will recognize and allow its use, even if it doesn't understand exactly how it works. (I see you have some mod morph behaviors defined [in your keymap](https://github.com/rambip/zmk-config/blob/master/config/reviung41.keymap#L44), if you're finding that the editor isn't letting you use them, feel free to comment here or create a new issue).

Thank you for reaching out! I'm glad you're getting some use out of the editor and I'm looking forward to including this enhancement at some point in the not too distant future :)

nickcoutsos, 2022-08-01:
I've been doing some research into the yaml schemas and started by writing a script to extract them all into something I can use for a generic behavior editor, but I don't think there's enough in the yaml alone to describe these (e.g., the `mods` property is just an integer, but nothing says that it should be a union of modifiers or what the enumeration of modifiers is: https://github.com/zmkfirmware/zmk/blob/main/app/include/dt-bindings/zmk/modifiers.h).

I think I'm still quite far from doing this as the mostly data-driven feature I'd hoped, but I could start work on this next and use it to guide me in implementing a pattern for other behaviors.

nickcoutsos, 2023-01-27:
Hi @rambip, it's been a while and I've made some progress on custom behaviors in-app. It's still pretty rough, but I've enabled the beta option for you if you wanted to take a look sometime. There are details about some of the limitations in the app and you opt not to use it if you'd rather wait but I just wanted to give you a heads up.

Add0z, 2023-01-27:
Oh great!! I'm waiting for this feature too!

nickcoutsos, 2023-01-27:
@Add0z thanks, I'll add you as well.

nickcoutsos, 2023-03-05:
Behavior editing (including mod morphs) is currently available for beta testing, feel free to create new issues if you run into problems.

closed 2023-03-05 by nickcoutsos