nickcoutsos/keymap-editor #40: Delete a macro makes the app crash

rambip, 2022-07-13:
I created a macro in my config, deleted it in the gui, and then tried to go bac to the main page of the editor.

I got this error:

```
TypeError

t is undefined

  keymap.js:7:19
  Keyboard/Binding/Binding.js:43:45
  ../node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  ../node_modules/react-dom/cjs/react-dom.production.min.js:289:385
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:391
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:322
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  ../node_modules/react-dom/cjs/react-dom.production.min.js:270:87
  ../node_modules/react-dom/cjs/react-dom.production.min.js:272:299
  ../node_modules/react-dom/cjs/react-dom.production.min.js:127:104
  ../node_modules/react-dom/cjs/react-dom.production.min.js:266:268
```

nickcoutsos, 2022-07-13:
Hey, thanks for reporting this!

If there's any more context you can provide it would help me to reproduce the error on my end:

- was that the only macro defined?
- had you committed your changes before deleting the macro? if so, did you commit changes after deleting it?
- what (if you can remember) did the macro contain?
    - binding/behavior type
    - number of binds

Was it a freak occurrence or is this something you've been able to reproduce more than once?

I'll keep playing around with it and let you know if I come up with anything.

nickcoutsos, 2022-07-13:
I just realized I'm probably overthinking this.

Do these steps sound like what happened to you?

1. Create macro
2. Bind that macro to a key in the keymap
3. Delete that macro
4. Return to keymap tab

nickcoutsos, 2022-07-14:
Just pushed up a change to address this.

It uses the same pattern as keycodes where a bind behaviour it doesn't recognize is displayed as `⦸`, highlighted in red, and a ⚠️ icon is displayed in the corner of that key. This prevents the crash from happening but it isn't treated as a critical error, either.

Nothing in the app will stop the user from making (other) changes to the keymap and saving them because it's still technically possible that the behaviour is valid but defined somewhere the editor can't discover. Easy examples of this are keymaps that `#include <blah.dtsi>` macros and behaviours shared amongst multiple keyboards. More complicated examples are keyboards that are building against a different branch/fork of the ZMK firmware repo that may have new or experimental features.

I'm not super pleased with this yet but no crashing is good and so is giving people flexibility. Eventually I think it would be good to collect all of these warnings in one place so that it's easier to review before committing changes.

I'll close the issue for now but feel free to re-open it if you'd like to have more of a discussion (or if I've completely overlooked something else!).

closed 2022-07-14 by nickcoutsos