nickcoutsos/keymap-editor #43: Some unicode symbols not displayed properly

Dimentium, 2022-08-03:
Hi!

The characters for the arrows (left, up, down, right) and the DELETE button are not displaying properly.

[image]

I installed Quicksand font, checked in Safari and Firefox. 
The results are the same.

nickcoutsos, 2022-08-03:
You know, I've seen other people have this issue but haven't investigated it yet myself. Are you on Windows?

I had assumed that the Unicode glyphs I picked would work for everybody, but most likely my OS falling back to a font where they're present while yours has no alternative.

Recently I've made some changes to my keycode definitions and turned the symbol/icon mappings into overlays something like
- base keycode definitions (e.g. `"RIGHT_ARROW"`)
- unicode glyphs (whichever ones I picked out that seemed useful, e.g. `⮐`)
- Windows-specific font-awesome glyphs (right now that's just the Windows logo for the GUI key)
- macOS-specific font-awesome glyphs (for the cmd and option keys)

At some point I should replace at least the problematic unicode glyphs with font-awesome alternatives, but in the meantime there are settings you can use to hopefully avoid the situation of blank symbols:
[image]

nickcoutsos, 2022-08-03:
Ah, hell, I just noticed that the [delete icon](https://fontawesome.com/icons/delete-right?s=solid) isn't part of the font-awesome free set. That's a shame, I'll have to look for something else for that or else just show `DEL` again.

I suppose I could get ridiculous and have it render the backspace symbol flipped around...

Dimentium, 2022-08-04:
Im on MacOS.

also on Linux arrows displayed ok:
[image]

I have not Windows right now for check that :(

nickcoutsos, 2022-08-04:
I think the fact that we're both on macOS and getting different results is enough to convince me to do something. I did like those arrows but font-awesome has some that look pretty much the same so there's no reason now to switch to that. I can get a fix out later tonight.

nickcoutsos, 2022-08-05:
Pushed some changes.

Arrows are now using font-awesome, and I've cheekily implemented the delete icon by flipping the backspace icon around. I think the only remaining "exotic" unicode glyphs are for command, option, shift, and enter. The latter two seem to be rendering fine in your screenshot, so we're probably fine but let me know if you run into other issues!

closed 2022-08-05 by nickcoutsos

Dimentium, 2022-08-05:
Awesome!
Thank you!! :)

[image]

Dimentium, 2022-08-05:
> Pushed some changes.
>
> Arrows are now using font-awesome, and I've cheekily implemented the delete icon by flipping the backspace icon around. I think the only remaining "exotic" unicode glyphs are [...]

Note that there is still a bad symbol here in the settings:
[image]

nickcoutsos, 2022-08-12:
Thanks, I'll make a note to update that as well!