nickcoutsos/keymap-editor #44: Cannot read properties of undefined (reading 'bindings')

Avada-sao, 2022-08-12:
```
TypeError

Cannot read properties of undefined (reading 'bindings')

  Keyboard/Keyboard.js:81:45
  ../node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  ../node_modules/react-dom/cjs/react-dom.production.min.js:215:269
  ../node_modules/react-dom/cjs/react-dom.production.min.js:291:201
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:388
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:319
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  ../node_modules/react-dom/cjs/react-dom.production.min.js:270:87
  ../node_modules/react-dom/cjs/react-dom.production.min.js:267:428
  ../node_modules/scheduler/cjs/scheduler.production.min.js:13:202
```

nickcoutsos, 2022-08-12:
Hi @Avada-sao thanks for sharing! Is there a zmk config repo you can point me to and/or any steps to reproduce this?

I have some guesses, was this after switching between keymaps (source or repo/branch) or deleting a layer?

nickcoutsos, 2022-08-16:
I've added some error checking to display a prompt in case there's no layer to display in a keymap. Feel free to re-open if that's not resolving the issue for you.

closed 2022-08-16 by nickcoutsos