nickcoutsos/keymap-editor #45: Combos menu had text blocking board

fpsmilky, 2022-08-21:
not sure how many of the keyboard layouts have this issue but the contra layout has the guide text overlapping.
[image]

fpsmilky, 2022-08-21:
if this doesnt get fixed, not a big deal. it can just be edited away from the inspect menu

[image]

nickcoutsos, 2022-08-21:
Ah, nice catch, thanks for sharing! I suspect it's just the narrower width of the layout causing the text to wrap. That should be an easy fix, I'll have a look when I'm home later.

nickcoutsos, 2022-08-21:
I could see this being a fairly common issue for non-split keyboards as the automatically generated layout will tend to be that shape by default. If the layout isn't wide enough then you'd wind up with that prompt text wrapping and vertically overlapping more than I originally intended.

[image]

- Added a CSS transition on hover to make the prompt more transparent
- Added `pointer-events: none` to let mouse hover/click pass through to the combo layout

I think I'll leave further changes for another time when I revisit the component, but this should prevent situations where the prompt is blocking initial input selection.

closed 2022-08-21 by nickcoutsos

fpsmilky, 2022-08-22:
awesome buddy, this configurator helped me so much with zmk 🤣