nickcoutsos/keymap-editor #49: Change selected Repo in github app

Eoin-Murphy, 2022-09-27:
Hi there, maybe I'm being dense but is there a way to change which repo the github app is using?
I can't see one and even when I clear all local storage it just re-authenticates to the same repo?

Do I need to de-authorize on the repo side?

nickcoutsos, 2022-09-27:
Hi!

I think the app could be enhanced a little to make this part of the integration easier. If you've only got one repository associated with the app then that's all you'll see in the list, but you can add more by going into your app installation settings.

1. Visit https://github.com/settings/installations
2. Click the _Configure_ button next to _ZMK Keymap Editor_
3. In the _Repository Access_ section add more of your repositories to the list
    - saving here should redirect you back to the app and update the available repositories, but if not then you can try clearing local storage and doing the auth flow again (I haven't looked at this in a little while so my memory is hazy)
    - note that I'm pretty strict on the use of the _All repositories_ option. If you select this your installation will be suspended until you set it back to a selection of appropriate repositories. The alternative has left the backend integration bombarded with build notifications for irrelevant repositories in the past.

Hopefully I can find time in the near future to jump back in and streamline some of this, but let me know if you have any trouble with the above details!

Eoin-Murphy, 2022-09-29:
Hi Nick, fantastic! Thank you so much for the response. Solves my question perfectly.

closed 2022-09-29 by Eoin-Murphy