nickcoutsos/keymap-editor #52: Feature Request: "Raw Mode"

ankushg, 2022-09-28:
It would be great to have a way to use the UI to dump arbitrary text into the keymap file

Maybe one way to implement it is to have a placeholder `raw` "behavior" that takes a raw string as its argument and inlines that into the keymap? 

Whenever the parser doesn't know what a given behavior is, it can just use the placeholder behavior and allow the user to edit the text via the UI

(Related to https://github.com/nickcoutsos/keymap-editor/discussions/50#discussioncomment-3754829 )

nickcoutsos, 2022-12-01:
I think I'm working toward this a little bit. I'm trying to deal with user's custom behaviours that accept non-keycode bindings (e.g., a hold tap where both actions are layer-related).

What I've got so far is:

1. My manual definitions of the expected parameters for the built-in (and bindable!) ZMK behaviours. By "parameters" I mean specifying whether a binding cell is a keycode, a layer index, or a "command" (like `BT_SEL`) and so on. This is manual because the ZMK codebase doesn't currently include this detail in the schema for its behaviours.
2. Parsing custom behaviours defined in the keymap, looking for a `bindings` property that references known behaviours, then grabbing the defined list of parameters.

This was pretty much just thrown together and I'm hoping to test it enough to deploy it soon, but I'm missing fallbacks for behaviours that can't be resolved as described above. The current live version of this just assumes any unknown parameter type is a keycode, but that could become `raw` or `unknown` instead. The latter is useful because in addition to using the value as-is for displaying in the UI and updating the keymap I can also provide a visual cue so that the user knows at the very least that have to be careful, but potentially I could prompt them for what they think the parameter is so that the appropriate component can be selected.

I realize that's not exactly what you've described, though. For your use-case are you envisioning a large number of undefined behaviours, or would it be worth having the user dig into the keymap file initially to define those behaviours for the app to use? Of course this is all just early steps towards being able to customize behaviours within the app.

nickcoutsos, 2023-09-04:
Gonna go ahead and close this. I don't quite support entering "raw" binding code, though behavior management is quite fleshed out.

The reason I won't likely take it to the point of having that escape hatch is that to enable a lot of the core functionality (for example, if you rename a behavior's label all references to it need to be updated, if you re-arrange your keymap's layers the references to that layer index must be updated, etc) the app needs to be able to understand what it's dealing with.

I try to support unrecognized behaviors (such as those for pointing devices when people are building against forks of ZMK) and there's some work still to be done, but even here I distinguish between the behavior and its parameters, accepting individual raw values for the latter.

closed 2023-09-04 by nickcoutsos