nickcoutsos/keymap-editor #56: Missing support for modifier function keypresses such as `LC(Z)`?

Richiban, 2022-10-07:
I may be missing something, but there appears to be no way to enter a keypress code such as `LC(Z)` into the UI. Funnily enough, if I enter a modifier function in my keymap file manually and then tell the UI to refresh from GitHub then I am able to _edit_ the keypress.

Am I missing something? This scenario isn't mentioned in the docs and I wasn't able to get it to work by exploring the UI.

Dimentium, 2022-10-07:
https://github.com/nickcoutsos/keymap-editor/discussions/50 ?

Dimentium, 2022-10-07:
https://github.com/nickcoutsos/keymap-editor/discussions/54#discussioncomment-3794398 ?

nickcoutsos, 2022-10-08:
@Dimentium shows how its done. In my implementation `LC` like `LCTRL` and others are bundled together as "keycodes", but `LC` (and other modifiers) are actually expressed as macro functions that accept another "keycode" as a parameter so when they're selected in the GUI you get to select another value.

This implementation was easy to reason about behind the scenes but isn't a great user experience and is something I'm giving some thought to improving, so thank you for sharing your experience!

closed 2022-10-08 by nickcoutsos

Richiban, 2022-10-09:
Many thanks for your input everyone. This has sorted me out!