nickcoutsos/keymap-editor #61: Encoders appear to have been defined backwards

Richiban, 2022-10-10:
It appears that the UI gets _clockwise_ and _anti-clockwise_ the wrong way round. See screenshot for how I've had to define my encoders incorrectly int order to get the keycodes to show up correctly in the keymap file:

[image]

The above produces the following keymap:

[image]

Richiban, 2022-10-10:
On having a second look, it appear they may have wired my encoders up backwards! The ZMK docs do say that the CW goes first, then the CCW key (so, in my case, it should be `<&inc_dec_kp DOWN UP &inc_dec_kp LA(RIGHT) LA(LEFT)>` instead. Might be helpful if someone else can confirm this!

nickcoutsos, 2022-10-10:
You had me a little worried there! I do indeed swap the widget order, displaying the decrement binding first, but it _should_ be mapping to the correct binding cell in the devicetree keymap. I would double-check this myself, but I don't have any encoders and all of my keyboards are hand-wired.

I'll leave this open for now if you're or anyone else can add more details!

closed 2022-12-01 by nickcoutsos