nickcoutsos/keymap-editor #65: Unable to use https://nickcoutsos.github.io/keymap-editor/

pujvz, 2022-10-22:
Whenever I click Add Repositories, I get a "Unicorn!" error from github.

[image]

```
This page is taking too long to load.

Sorry about that. Please try refreshing and contact us if the problem persists.
```

nickcoutsos, 2022-10-22:
Oh no, I'm really sorry to hear that! Whenabouts was this? I don't see errors logged about this but I've also never encountered a message like that. This was on GitHub's website?

pujvz, 2022-10-22:
Yes, this is on github. I encountered this earlier today and am still experiencing the same error. Can you please try reproducing the error by visiting this link? https://github.com/apps/zmk-keymap-editor

nickcoutsos, 2022-10-23:
They seem to work alright for me, I just deleted my app installation and went through the process again. Just to be clear, are you getting this error when going to the new installation page or after selecting repositories to install?

In the GitHub logs I see you revoking app authorization (the login part) but nothing about the installation getting created.

pujvz, 2022-10-23:
I am getting the same error after visiting both links above.
Sorry, I am a bit new to github, so I am not sure if the app is installed. I tried revoking access and restarting the process after getting the error.

I followed the instructions in your readme's Setup > Web section.

These were the steps I took:
1. I cloned a repo with keymap configs for my keyboard.
2. went to https://nickcoutsos.github.io/keymap-editor/ and set Source to GitHub.
3. authorized the app. 
4. clicked Add Repository.
5. received the error.

nickcoutsos, 2022-10-23:
This is really puzzling! I've been watching the logs and I'm not even seeing GitHub call back to my app when you hit the installation page so I don't know what could be happening internally.

pujvz, 2022-10-23:
I found a solution in the community help discussions. I don't know why this was causing the error but after removing my organizations I was able to load the page without error:
https://github.com/orgs/community/discussions/35026#discussioncomment-3878922

nickcoutsos, 2022-10-23:
That's great! I saw another discussion where that suggestion was made but I'm surprised that it isn't something GitHub accounts for. I also wonder how many is too many for organization repositories... evidently the number of repos I access for work aren't taxing enough :P

I'm going to close this now, but re-open if you run into a related problem. I suppose once you've added your own repo (or, probably, once the app is installed for your account) you should be ok to re-add your organization, but if that still causes problems I'd be interested in knowing.

closed 2022-10-23 by nickcoutsos

pujvz, 2022-10-23:
Thanks for taking the time to look into this issue and for making this software! It's very intuitive and easy to use :).