nickcoutsos/keymap-editor #66: Feature Request: Support for arbitrary key codes

pujvz, 2022-10-23:
My .keymap file has a custom binding defined (&mo_sl). However I am unable to input a layer number as the key code value. For example, it auto changes (layer) 1 to N1. It would be very convenient to have the ability to override suggested keycodes and input arbitrary values.
[image]
[image]

Loving this software so far :smile: . Being able to visualize the layout has sped up configuration considerably.

nickcoutsos, 2022-10-23:
Thank you @pujvz 

More flexibility to handle custom behaviours (or rathr handling behaviours generically) is on my radar but its very tricky.

Based on your description I would guess (by the way, I'm personally not that experienced with configuring new behaviours myself) that `mo_sl` is using `compatible = "zmk,behavior-hold-tap";` and `bindings = <&mo, &sl>;`? Hold-tap seems like a popular behaviour to customize so it might even be valuable in the short term to add specific handling for it, but ultimately I'd love to see ZMK's behaviour schema fleshed out to describe what kinds of values are expected in `bindings` so that the editor can support it without explicitly knowing how its supposed to work.

There actually is some minimal support for handling custom behaviours right now, but it doesn't go as far as checking what parameter types are used, treating them all as keycodes instead.

The path forward would mean looking at `&mo, &sl` and resolving them to the momentary layer and sticky layer behaviours, and then using their `bindings` properties to come up with `layer` and `layer`. That last step, unfortunately, is based on my manual interpretation of the documentation. `&mo` is described in [app/dts/behaviors/momentary_layer.dtsi](https://github.com/zmkfirmware/zmk/blob/main/app/dts/behaviors/momentary_layer.dtsi) and [app/dts/bindings/behaviors/zmk,behavior-momentary-layer.yaml](https://github.com/zmkfirmware/zmk/blob/main/app/dts/bindings/behaviors/zmk%2Cbehavior-momentary-layer.yaml) in the ZMK codebase but they only tell you that the binding should have one parameter, and not that the parameter should be a layer index.

pujvz, 2022-10-24:
> mo_sl is using compatible = "zmk,behavior-hold-tap"; and bindings = <&mo, &sl>
That is correct. And I agree that zmk's documentation could definitely be clearer. I had to dig through the docs and ask them a question to figure out how to create `mo_sl`.

Thanks for explaining why this tricky. Honestly, your app is already in a nice working state. It accomplishes 90% of the things that I need to do when configuring zmk. The remaining 10% can be achieved with some text editing. Feel free to sit on this on request for a while or close it if you don't have the bandwidth. Thanks again for this app!

nickcoutsos, 2022-10-24:
It's great having another use case to reference! I'm going to leave this open for now, but I'd like to find some time to collect my thoughts about this in one place and reference the other issues requesting this kind of feature.

nickcoutsos, 2023-03-03:
Hi @pujvz, I realize I never got back to you on this after the work I've done!

Some time ago I added functionality to parse behaviors and resolve value types for its parameters. This should take care of the use case you described -- thank you again for bringing it up, it really helped me get started on some significant work. From there I went on to add in support for creating and editing behaviors from within the app, which is currently in a "closed beta". If you're still using the app and would like to give it a try, just let me know!

closed 2023-03-03 by nickcoutsos

pujvz, 2023-03-23:
Awesome!
Glad to see you continuing to improve your app :smiley: . I haven't used it in a while since my layout was working well. However I will be tinkering with it more over the next few days.