nickcoutsos/keymap-editor #67: Keymap-editor stuck on Waiting for API..

Designbykivi, 2022-10-25:
Heya, tried opening up the [web keymap-editor](https://nickcoutsos.github.io/keymap-editor/), but it just says "Waiting for API..". Tried several browsers and even on my phone, but the issues persists.

Alternatively I tried to set up the editor locally by following the [readme](https://github.com/nickcoutsos/keymap-editor/blob/main/running-locally.md), but eventually got stuck on **step 4**: `npm run dev`.  It errored out with the following message:
```
> keymap-editor@1.0.0 dev
> cross-env ENABLE_DEV_SERVER=true node index.js

node:internal/url:565
  throw new ERR_INVALID_URL(input);
  ^

TypeError [ERR_INVALID_URL]: Invalid URL
    at new NodeError (node:internal/errors:387:5)
    at URL.onParseError (node:internal/url:565:9)
    at new URL (node:internal/url:641:5)
    at Object.<anonymous> (/Users/kivi/Downloads/ZMK/keymap-editor/api/index.js:12:20)
    at Module._compile (node:internal/modules/cjs/loader:1155:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1209:10)
    at Module.load (node:internal/modules/cjs/loader:1033:32)
    at Function.Module._load (node:internal/modules/cjs/loader:868:12)
    at Module.require (node:internal/modules/cjs/loader:1057:19)
    at require (node:internal/modules/cjs/helpers:103:18) {
  input: 'undefined',
  code: 'ERR_INVALID_URL'
}
```

nickcoutsos, 2022-10-25:
Thanks for the heads up, that's really weird. Something seems to be up with the provider, I'll see what I can do once I'm home from work and let you know.

nickcoutsos, 2022-10-25:
I haven't done anything but it looks like it successfully started itself back up... hopefully it stays that way, there haven't been any code or configuration changes so if it's something on the provider's side I haven't seen them confirm it.

closed 2022-11-20 by nickcoutsos