nickcoutsos/keymap-editor #68: keymap editor crash

xaelectronics, 2022-11-26:
Crash has occurred twice, both times when I click on a key to edit it.

```
TypeError

g is undefined

  EnhancedPickers/Keycode/KeycodePicker.js:60:7
  ../node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  ../node_modules/react-dom/cjs/react-dom.production.min.js:289:385
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:391
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:322
  ../node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  ../node_modules/react-dom/cjs/react-dom.production.min.js:270:87
  ../node_modules/react-dom/cjs/react-dom.production.min.js:272:299
  ../node_modules/react-dom/cjs/react-dom.production.min.js:127:104
  ../node_modules/react-dom/cjs/react-dom.production.min.js:266:268
```

nickcoutsos, 2022-11-26:
Thanks for the heads up! I think I know what the issue is, I'll have a look and let you know here when a fix is deployed.

nickcoutsos, 2022-11-27:
@xaelectronics I've updated the new picker to better handle undefined keycodes, give it another shot when you have time and let me know if it goes :)

xaelectronics, 2022-11-30:
Seems to be resolved, great work!

closed 2022-11-30 by xaelectronics