nickcoutsos/keymap-editor #69: Keymap fails to load, error handler also doesn't load

xaelectronics, 2022-12-02:
The keymap editor fails to load for my heavily customized keymap. Default keymaps load fine. The customized keymap has a fair number of custom behaviors, including mouse behaviors from the ZMK mouse button fork.

```
GEThttps://zmk-keymap-editor.fly.dev/github/keyboard-files/29762192/xaelectronics/zmk_swept_kb_config?branch=miryoku&useDevicetree=true
[HTTP/2 500 Internal Server Error 1031ms]
```

```
Uncaught (in promise) Error: Request failed with status code 500
    exports createError.js:16
    exports settle.js:17
    w xhr.js:66
    exports xhr.js:78
    exports xhr.js:15
    exports dispatchRequest.js:58
    request Axios.js:108
    exports bind.js:9
    e api.js:33
    c runtime.js:63
    _invoke runtime.js:294
    k runtime.js:119
    Babel 4
    value api.js:8
    e api.js:211
    c runtime.js:63
    _invoke runtime.js:294
    k runtime.js:119
    Babel 4
    value api.js:8
    e Picker.js:97
    c runtime.js:63
    _invoke runtime.js:294
    k runtime.js:119
    Babel 4
    ef Picker.js:205
    React 3
    w scheduler.production.min.js:13
    M scheduler.production.min.js:14
    6813 scheduler.production.min.js:14
    Webpack 12
[createError.js:16:14](https://nickcoutsos.github.io/keymap-editor/static/node_modules/axios/lib/core/createError.js)
    Babel 6
    ef Picker.js:205
    React 3
    w scheduler.production.min.js:13
    M scheduler.production.min.js:14
    (Async: EventHandlerNonNull)
    6813 scheduler.production.min.js:14
    Webpack 12
```

nickcoutsos, 2022-12-02:
Sorry about this! It sounds like the main issue is the unrecognized mouse keys behaviour. I'm nearing completion of some work to better handle this stuff but it looks like I've jumped the gun on deploying the in-progress work. I'll try to roll back that deploy shortly.

nickcoutsos, 2022-12-02:
I'm trying to reproduce the error (for testing purposes, I'm still going to roll back) by loading undefined behaviors in my own keymap but it isn't happening. Are there any details about your keymap you'd be able to share?

nickcoutsos, 2022-12-02:
I caught your error in the server logs this time! I'll be able to work from that and ensure I'm rolling back to a working version.

xaelectronics, 2022-12-02:
Ah, I just saw this. I added you as a collaborator to the repo that I use for my keymap. It's the miryouku branch that's causing issues.

nickcoutsos, 2022-12-03:
Thanks @xaelectronics, I'm not sure I'm able to add your repo to my installation (maybe if I fork it, but I'd rather minimize contact with your private code). Could you test again or let me know if the problems have been resolved? The extensive set of custom behaviors have been really interesting to look at, though and should help me as I build out this feature a little more carefully!