nickcoutsos/keymap-editor #70: Automatic Layout Generation Failed

magagne, 2022-12-03:
The default .keymap generated file for a "splitkb.com Aurora Sweep" with "nice!nano v2" do not want to generate the layout. Safari browser, Ventura 13.01.

nickcoutsos, 2022-12-03:
Thanks, I haven't been keeping up with new boards. The metadata file has been updated, you can give it another shot now.

magagne, 2022-12-03:
Yahoo! it load now, but I have an error at the bottom of GUI screen (ref attachment) any thought ?

Sensor bindings
L_ENCODER1  R_ENCODER1
⦸                       ⦸

Cheers,
Matthieu .

[image]

nickcoutsos, 2022-12-04:
If you hover over the caution icon it should tell you, but I think it's just complaining that there's no sensor binding (or alternatively, not one that is recognized).

magagne, 2022-12-08:
"are you not expecting to see the encoder widgets at all?"
Exactly. and I decide to just ignored it, and in the config file it do not show anything about sensor though. So do not understand where that come from, maybe just the web app.

nickcoutsos, 2022-12-08:
I'm referring to [`splitkb_aurora_sweep.dtsi`](https://github.com/zmkfirmware/zmk/blob/001105e31844f0f59b82a2e71a4c3ada980b931b/app/boards/shields/splitkb_aurora_sweep/splitkb_aurora_sweep.dtsi#L58).

In this commit https://github.com/magagne/zmk-config/commit/1e7b4302af299787fbb174a4ffede9ea35130c8a I think you would need to have that `sensors {...}` node _inside_ the root node, like

```
/ {
    sensors {
        ...
    };

    keymap {
        ...
    };
};
```

Failing that, you can copy the `boards/shields/splitkb_aurora_sweep` directory structure (place it in the `config/` directory in your repo) or if it's not impacting the build maybe just ignore it.

closed 2022-12-21 by nickcoutsos

KatharinaSick, 2022-12-28:
just fyi: I am facing the same issue and tried to remove the sensors by adding an empty sensors block to the .keymap file (see [this commit](https://github.com/KatharinaSick/aurora-sweep-zmk-config/commit/c19109f42fb0a7cbe5146e3842a8a6b13a98ff7c)). Unfortunately, it didn't help. But it's not a big issue for me - I'm ignoring it, I just wanted to let you know :) 

Btw, I adjusted the default [info.json](https://github.com/KatharinaSick/aurora-sweep-zmk-config/blob/main/config/info.json) to match the Aurora Sweep better (it's not perfect but much closer to the original). Feel free to copy this one for the Aurora/Ferris Sweep if you want to :)

magagne, 2022-12-28:
Few weeks ago those sensors disappear (to my happy surprise) but for no reason they came back again, just ignore them, they will maybe disappear again ;-)

nickcoutsos, 2022-12-28:
The node would need to be `sensors { compatible = "zmk,sensors"; ... };` but apparently the editor just isn't set up to look for it in the keymap file, sadly. At some point I'll need to revisit sensors and get it to local modifications to the individual encoder nodes.

magagne, 2022-12-28:
> Btw, I adjusted the default [info.json](https://github.com/KatharinaSick/aurora-sweep-zmk-config/blob/main/config/info.json) to match the Aurora Sweep better (it's not perfect but much closer to the original). Feel free to copy this one for the Aurora/Ferris Sweep if you want to :)

Thanks I looked at it, nice rotation on thumbs button.

You can grab mine too as a good proportional staggered keys layout [info.json](https://github.com/magagne/zmk-config/blob/9bb2e6c025102f57b4297bb670799c5576a31943/config/info.json) only thing I need to resolve is the rotation of the thumbs button. So complicated to have a nice result, I just abandoned :.-(

nickcoutsos, 2022-12-29:
> So complicated to have a nice result, I just abandoned :.-(

Sorry about this. I was looking at it a while ago because I noticed things not working out that nicely myself. I think it came down to an inappropriate offset being applied in the rotation calculations. Would you be interested in creating a separate issue for this one? I wouldn't mind having some more user examples to test on.

magagne, 2022-12-29:
Yes will do.

tkx nick.

magagne, 2022-12-29:
> Yes will do. 
issu #76

nickcoutsos, 2023-04-26:
@magagne @KatharinaSick not a complete solution to the encoder issue yet, but I've taken to adding a `sensors` array to the metadata JSON to avoid unnecessary `.dts`/`.dtsi`/`.overlay` lookups whenever you load your keymap. The effect of this is that you can set this as `"sensors": []` to indicate to the editor that there are no sensors to manage.