nickcoutsos/keymap-editor #72: sticky key do not show again

magagne, 2022-12-16:
hi there,

the problem came back again about the sticky key, I cannot do configuration in the UI.

[image]

[image]

cheers.
Matthieu.

nickcoutsos, 2022-12-16:
Sorry about that, I've been trying to merge in some work that's been sitting around for a while. I've deployed a fix.

magagne, 2022-12-16:
lol regression bug...that can happen to anybody do not be sorry ;-)

closed 2022-12-21 by nickcoutsos