nickcoutsos/keymap-editor #73: Feature Request: Tap dance

wvffle, 2022-12-16:
I'd love to be able to define custom tap dances straight from the UI

nickcoutsos, 2022-12-16:
Hi @wvffle 

Thanks for the feature request. Support editing behaviours is definitely on my radar but there are some complexities around it.

I'm making some baby steps towards that goal with parsing of ZMK behaviour definitions and have recently started reworking some abstractions on devicetree handling in my codebase that should help me towards this. I don't have a timeline for when I'll be able to get to this but hopefully over the holidays I can at least get a little more organized and use GitHub's projects to track milestones.

nickcoutsos, 2023-01-23:
Not ready yet, but just want to share that this is in progress and I'm using #78 as a blog of sorts to share updates. Hope to make this available for beta testing in the coming weeks.

nickcoutsos, 2023-01-27:
Hi @wvffle, it's still rough around the edges, but I've enabled the beta behavior editing for you. It's an opt-in if you'd rather wait but next time you're using the keymap editor you can have a look at some of the details.

jackytrabbit, 2023-02-26:
@nickcoutsos Hi Nick, I have been following your work for a while. I want to say thank you for your hard work and the keymap editor is really a great tool for ZMK users like me! I am looking forward to Tap Dance feature as a small keyboard user. But take your time, you have already done so much for us!

nickcoutsos, 2023-02-26:
Hi @jackytrabbit, thanks for chiming in and sharing your interest! What I really need right now is volunteers to beta test and make sure I'm not overlooking things due to my limited personal use cases. I've added you to the list now so if you're interested you can enable the behavior editing beta next time you're in the app and try it out.

jackytrabbit, 2023-02-27:
> 

Thanks for adding me to the beta test! I will try those features out!

If you need volunteers, you can add @Mengsicheng to the beta test. He is using a zmk keyboard and want to try Tap Dance feature in your app too. :P

jamiemoore, 2023-03-01:
@nickcoutsos I would be happy to help test too.

nickcoutsos, 2023-03-01:
Thanks @jamiemoore, I've added you as well!

nickcoutsos, 2023-03-04:
I'm going to go ahead and close this. I haven't had much feedback from the "closed beta" so I've just opened it up to all users.

closed 2023-03-04 by nickcoutsos