nickcoutsos/keymap-editor #74: Bug: something is breaking the github commit feature for me

tttaaannnggg, 2022-12-23:
Heyyo, I'm trying to do a commit to a branch on my gh fork and am getting 500 internal service errors. Of note - the 'commit' button is stuck with a spinner

[image]

I've been able to open the same branch in a new window and commit successfully, so i'm not sure what's going on here (guessing it's either some kind of session issue or something to do with my layout itself). I captured the request here as a curl, though for obvious reasons i have redacted my auth token:

request:
```
curl 'https://zmk-keymap-editor.fly.dev/github/keyboard-files/32479307/tttaaannnggg%2FAdv360-Pro-ZMK/wip' \
  -H 'authority: zmk-keymap-editor.fly.dev' \
  -H 'accept: application/json, text/plain, */*' \
  -H 'accept-language: en-US,en;q=0.9,es;q=0.8' \
  -H 'authorization: REDACTED' \
  -H 'content-type: application/json' \
  -H 'dnt: 1' \
  -H 'origin: https://nickcoutsos.github.io' \
  -H 'referer: https://nickcoutsos.github.io/' \
  -H 'sec-ch-ua: " Not A;Brand";v="99", "Chromium";v="100", "Google Chrome";v="100"' \
  -H 'sec-ch-ua-mobile: ?0' \
  -H 'sec-ch-ua-platform: "macOS"' \
  -H 'sec-fetch-dest: empty' \
  -H 'sec-fetch-mode: cors' \
  -H 'sec-fetch-site: cross-site' \
  -H 'user-agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/100.0.4896.88 Safari/537.36' \
  --data-raw $'{"layout":[{"label":"=","row":0,"col":0,"x":0,"y":0.25,"w":1.25},{"label":"1","row":0,"col":1,"x":1.25,"y":0.25},{"label":"2","row":0,"col":2,"x":2.25,"y":0},{"label":"3","row":0,"col":3,"x":3.25,"y":0},{"label":"4","row":0,"col":4,"x":4.25,"y":0},{"label":"5","row":0,"col":5,"x":5.25,"y":0},{"label":"mod1","row":0,"col":6,"x":6.25,"y":0},{"label":"mod2","row":0,"col":7,"x":10.75,"y":0},{"label":"6","row":0,"col":8,"x":11.75,"y":0},{"label":"7","row":0,"col":9,"x":12.75,"y":0},{"label":"8","row":0,"col":10,"x":13.75,"y":0},{"label":"9","row":0,"col":11,"x":14.75,"y":0},{"label":"0","row":0,"col":12,"x":15.75,"y":0.25},{"label":"-","row":0,"col":13,"x":16.75,"y":0.25,"w":1.25},{"label":"Tab","row":1,"col":0,"x":0,"y":1.25,"w":1.25},{"label":"Q","row":1,"col":1,"x":1.25,"y":1.25},{"label":"W","row":1,"col":2,"x":2.25,"y":1},{"label":"E","row":1,"col":3,"x":3.25,"y":1},{"label":"R","row":1,"col":4,"x":4.25,"y":1},{"label":"T","row":1,"col":5,"x":5.25,"y":1},{"label":"mod3","row":1,"col":6,"x":6.25,"y":1},{"label":"mod4","row":1,"col":7,"x":10.75,"y":1},{"label":"Y","row":1,"col":8,"x":11.75,"y":1},{"label":"U","row":1,"col":9,"x":12.75,"y":1},{"label":"I","row":1,"col":10,"x":13.75,"y":1},{"label":"O","row":1,"col":11,"x":14.75,"y":1},{"label":"P","row":1,"col":12,"x":15.75,"y":1.25},{"label":"\\\\","row":1,"col":13,"x":16.75,"y":1.25,"w":1.25},{"label":"Caps","row":2,"col":0,"x":0,"y":2.25,"w":1.25},{"label":"A","row":2,"col":1,"x":1.25,"y":2.25},{"label":"S","row":2,"col":2,"x":2.25,"y":2},{"label":"D","row":2,"col":3,"x":3.25,"y":2},{"label":"F","row":2,"col":4,"x":4.25,"y":2},{"label":"G","row":2,"col":5,"x":5.25,"y":2},{"label":"mod5","row":2,"col":6,"x":6.25,"y":2},{"label":"LCtrl","row":2,"col":7,"x":6.75,"y":4,"r":15,"rx":5.25,"ry":4},{"label":"LAlt","row":2,"col":8,"x":7.75,"y":4,"r":15,"rx":5.25,"ry":4},{"label":"LGui","row":2,"col":11,"x":9.25,"y":4,"r":-15,"rx":12.75,"ry":4},{"label":"RCtrl","row":2,"col":12,"x":10.25,"y":4,"r":-15,"rx":12.75,"ry":4},{"label":"mod6","row":2,"col":13,"x":10.75,"y":2},{"label":"H","row":2,"col":14,"x":11.75,"y":2},{"label":"J","row":2,"col":15,"x":12.75,"y":2},{"label":"K","row":2,"col":16,"x":13.75,"y":2},{"label":"L","row":2,"col":17,"x":14.75,"y":2},{"label":";","row":2,"col":18,"x":15.75,"y":2.25},{"label":"\'","row":2,"col":19,"x":16.75,"y":2.25,"w":1.25},{"label":"LShift","row":3,"col":0,"x":0,"y":3.25,"w":1.25},{"label":"Z","row":3,"col":1,"x":1.25,"y":3.25},{"label":"X","row":3,"col":2,"x":2.25,"y":3},{"label":"C","row":3,"col":3,"x":3.25,"y":3},{"label":"V","row":3,"col":4,"x":4.25,"y":3},{"label":"B","row":3,"col":5,"x":5.25,"y":3},{"label":"1U1","row":3,"col":6,"x":5.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":0},{"label":"1U2","row":3,"col":7,"x":6.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":0},{"label":"Home","row":3,"col":8,"x":7.75,"y":5,"r":15,"rx":5.25,"ry":4},{"label":"Ped3","row":3,"col":9,"x":20,"y":5},{"label":"Ped4","row":3,"col":10,"x":21,"y":5},{"label":"PgUp","row":3,"col":11,"x":9.25,"y":5,"r":-15,"rx":12.75,"ry":4},{"label":"1U3","row":3,"col":12,"x":10.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":0},{"label":"1U4","row":3,"col":13,"x":11.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":0},{"label":"N","row":3,"col":14,"x":11.75,"y":3},{"label":"M","row":3,"col":15,"x":12.75,"y":3},{"label":",","row":3,"col":16,"x":13.75,"y":3},{"label":".","row":3,"col":17,"x":14.75,"y":3},{"label":"/","row":3,"col":18,"x":15.75,"y":3.25},{"label":"RShift","row":3,"col":19,"x":16.75,"y":3.25,"w":1.25},{"label":"mod7","row":4,"col":0,"x":0,"y":4.25,"w":1.25},{"label":"`","row":4,"col":1,"x":1.25,"y":4.25},{"label":"\\\\","row":4,"col":2,"x":2.25,"y":4},{"label":"V","row":4,"col":3,"x":3.25,"y":4},{"label":"B","row":4,"col":4,"x":4.25,"y":4},{"label":"Bksp","row":4,"col":5,"x":5.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":2},{"label":"Del","row":4,"col":6,"x":6.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":2},{"label":"End","row":4,"col":7,"x":7.75,"y":6,"r":15,"rx":5.25,"ry":4},{"label":"PgDn","row":4,"col":8,"x":9.25,"y":6,"r":-15,"rx":12.75,"ry":4},{"label":"Enter","row":4,"col":9,"x":10.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":2},{"label":"Space","row":4,"col":10,"x":11.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":2},{"label":"Up","row":4,"col":11,"x":12.75,"y":4},{"label":"Down","row":4,"col":12,"x":13.75,"y":4},{"label":",","row":4,"col":13,"x":14.75,"y":4},{"label":".","row":4,"col":14,"x":15.75,"y":4.25},{"label":"mod8","row":4,"col":15,"x":16.75,"y":4.25,"w":1.25}],"keymap":{"keyboard":"foo","keymap":"","layout":"","behaviors":[{"binding":"&hm","code":"&hm","aliases":["&hm"],"identifier":"homerow_mods","bindingCells":2,"label":"HOMEROW_MODS","compatible":"zmk,behavior-hold-tap","bindings":["&kp","&kp"],"description":"Hold or Tap behavior","include":"two_param.yaml","properties":{"bindings":{"type":"phandles","required":true},"tapping-term-ms":{"type":"int"},"tapping_term_ms":{"type":"int"},"quick-tap-ms":{"type":"int","default":-1},"quick_tap_ms":{"type":"int"},"global-quick-tap":{"type":"boolean"},"flavor":{"type":"string","required":false,"default":"hold-preferred","enum":["hold-preferred","balanced","tap-preferred","tap-unless-interrupted"]},"retro-tap":{"type":"boolean"},"hold-trigger-key-positions":{"type":"array","required":false,"default":[]},"#binding-cells":{"type":"int","required":true,"const":2},"label":{"type":"string","required":true}},"params":["code","code"]}],"defines":{},"layers":[{"name":"default_layer","identifier":"default_layer","bindings":[{"value":"&kp","params":[{"value":"GRAVE","slot":"code","params":[]}]},{"value":"&kp","params":[{"value":"N1","params":[]}]},{"value":"&kp","params":[{"value":"N2","params":[]}]},{"value":"&kp","params":[{"value":"N3","params":[]}]},{"value":"&kp","params":[{"value":"N4","params":[]}]},{"value":"&kp","params":[{"value":"N5","params":[]}]},{"value":"&none","params":[]},{"value":"&none","params":[]},{"value":"&kp","params":[{"value":"N6","params":[]}]},{"value":"&kp","params":[{"value":"N7","params":[]}]},{"value":"&kp","params":[{"value":"N8","params":[]}]},{"value":"&kp","params":[{"value":"N9","params":[]}]},{"value":"&kp","params":[{"value":"N0","params":[]}]},{"value":"&kp","params":[{"value":"MINUS","params":[]}]},{"value":"&kp","params":[{"value":"TAB","params":[]}]},{"value":"&kp","params":[{"value":"Q","params":[]}]},{"value":"&kp","params":[{"value":"W","params":[]}]},{"value":"&kp","params":[{"value":"E","params":[]}]},{"value":"&kp","params":[{"value":"R","params":[]}]},{"value":"&kp","params":[{"value":"T","params":[]}]},{"value":"&none","params":[]},{"value":"&none","params":[]},{"value":"&kp","params":[{"value":"Y","params":[]}]},{"value":"&kp","params":[{"value":"U","params":[]}]},{"value":"&kp","params":[{"value":"I","params":[]}]},{"value":"&kp","params":[{"value":"O","params":[]}]},{"value":"&kp","params":[{"value":"P","params":[]}]},{"value":"&kp","params":[{"value":"BACKSPACE","slot":"code","params":[]}]},{"value":"&lt","params":[{"value":1,"slot":"layer","params":[]},{"value":"ESC","slot":"code","params":[]}]},{"value":"&kp","params":[{"value":"A","params":[]}]},{"value":"&kp","params":[{"value":"S","params":[]}]},{"value":"&kp","params":[{"value" [... cut ...] ams":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&trans","params":[]},{"value":"&bootloader","params":[]},{"value":"&trans","params":[]}]}],"combos":[],"macros":[],"layer_names":[]},"message":"pls"}' \
  --compressed
```

nickcoutsos, 2022-12-23:
Hi there

The idea of a session issue kind of captures the problem.

> I've been able to open the same branch in a new window and commit successfully,

The keymap editor tracks list indices of layer elements as they change between reading and writing, so by attempting to make changes in another session you can end up in a situation where you're trying to save changes to layers that reference indices that aren't present anymore. It's a bit complicated but that complexity is needed to preserve properties and other feature not managed by the editor (comments or labels for example).

I'm very sorry for the inconvenience. If there's a lot of work in that failed request that you don't want to lose you can try the following payload (you may have to refresh the editor and inspect some network requests to get a fresh auth token)

```
{"layout":[{"label":"=","row":0,"col":0,"x":0,"y":0.25,"w":1.25},{"label":"1","row":0,"col":1,"x":1.25,"y":0.25},{"label":"2","row":0,"col":2,"x":2.25,"y":0},{"label":"3","row":0,"col":3,"x":3.25,"y":0},{"label":"4","row":0,"col":4,"x":4.25,"y":0},{"label":"5","row":0,"col":5,"x":5.25,"y":0},{"label":"mod1","row":0,"col":6,"x":6.25,"y":0},{"label":"mod2","row":0,"col":7,"x":10.75,"y":0},{"label":"6","row":0,"col":8,"x":11.75,"y":0},{"label":"7","row":0,"col":9,"x":12.75,"y":0},{"label":"8","row":0,"col":10,"x":13.75,"y":0},{"label":"9","row":0,"col":11,"x":14.75,"y":0},{"label":"0","row":0,"col":12,"x":15.75,"y":0.25},{"label":"-","row":0,"col":13,"x":16.75,"y":0.25,"w":1.25},{"label":"Tab","row":1,"col":0,"x":0,"y":1.25,"w":1.25},{"label":"Q","row":1,"col":1,"x":1.25,"y":1.25},{"label":"W","row":1,"col":2,"x":2.25,"y":1},{"label":"E","row":1,"col":3,"x":3.25,"y":1},{"label":"R","row":1,"col":4,"x":4.25,"y":1},{"label":"T","row":1,"col":5,"x":5.25,"y":1},{"label":"mod3","row":1,"col":6,"x":6.25,"y":1},{"label":"mod4","row":1,"col":7,"x":10.75,"y":1},{"label":"Y","row":1,"col":8,"x":11.75,"y":1},{"label":"U","row":1,"col":9,"x":12.75,"y":1},{"label":"I","row":1,"col":10,"x":13.75,"y":1},{"label":"O","row":1,"col":11,"x":14.75,"y":1},{"label":"P","row":1,"col":12,"x":15.75,"y":1.25},{"label":"\\\\","row":1,"col":13,"x":16.75,"y":1.25,"w":1.25},{"label":"Caps","row":2,"col":0,"x":0,"y":2.25,"w":1.25},{"label":"A","row":2,"col":1,"x":1.25,"y":2.25},{"label":"S","row":2,"col":2,"x":2.25,"y":2},{"label":"D","row":2,"col":3,"x":3.25,"y":2},{"label":"F","row":2,"col":4,"x":4.25,"y":2},{"label":"G","row":2,"col":5,"x":5.25,"y":2},{"label":"mod5","row":2,"col":6,"x":6.25,"y":2},{"label":"LCtrl","row":2,"col":7,"x":6.75,"y":4,"r":15,"rx":5.25,"ry":4},{"label":"LAlt","row":2,"col":8,"x":7.75,"y":4,"r":15,"rx":5.25,"ry":4},{"label":"LGui","row":2,"col":11,"x":9.25,"y":4,"r":-15,"rx":12.75,"ry":4},{"label":"RCtrl","row":2,"col":12,"x":10.25,"y":4,"r":-15,"rx":12.75,"ry":4},{"label":"mod6","row":2,"col":13,"x":10.75,"y":2},{"label":"H","row":2,"col":14,"x":11.75,"y":2},{"label":"J","row":2,"col":15,"x":12.75,"y":2},{"label":"K","row":2,"col":16,"x":13.75,"y":2},{"label":"L","row":2,"col":17,"x":14.75,"y":2},{"label":";","row":2,"col":18,"x":15.75,"y":2.25},{"label":"\'","row":2,"col":19,"x":16.75,"y":2.25,"w":1.25},{"label":"LShift","row":3,"col":0,"x":0,"y":3.25,"w":1.25},{"label":"Z","row":3,"col":1,"x":1.25,"y":3.25},{"label":"X","row":3,"col":2,"x":2.25,"y":3},{"label":"C","row":3,"col":3,"x":3.25,"y":3},{"label":"V","row":3,"col":4,"x":4.25,"y":3},{"label":"B","row":3,"col":5,"x":5.25,"y":3},{"label":"1U1","row":3,"col":6,"x":5.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":0},{"label":"1U2","row":3,"col":7,"x":6.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":0},{"label":"Home","row":3,"col":8,"x":7.75,"y":5,"r":15,"rx":5.25,"ry":4},{"label":"Ped3","row":3,"col":9,"x":20,"y":5},{"label":"Ped4","row":3,"col":10,"x":21,"y":5},{"label":"PgUp","row":3,"col":11,"x":9.25,"y":5,"r":-15,"rx":12.75,"ry":4},{"label":"1U3","row":3,"col":12,"x":10.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":0},{"label":"1U4","row":3,"col":13,"x":11.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":0},{"label":"N","row":3,"col":14,"x":11.75,"y":3},{"label":"M","row":3,"col":15,"x":12.75,"y":3},{"label":",","row":3,"col":16,"x":13.75,"y":3},{"label":".","row":3,"col":17,"x":14.75,"y":3},{"label":"/","row":3,"col":18,"x":15.75,"y":3.25},{"label":"RShift","row":3,"col":19,"x":16.75,"y":3.25,"w":1.25},{"label":"mod7","row":4,"col":0,"x":0,"y":4.25,"w":1.25},{"label":"`","row":4,"col":1,"x":1.25,"y":4.25},{"label":"\\\\","row":4,"col":2,"x":2.25,"y":4},{"label":"V","row":4,"col":3,"x":3.25,"y":4},{"label":"B","row":4,"col":4,"x":4.25,"y":4},{"label":"Bksp","row":4,"col":5,"x":5.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":2},{"label":"Del","row":4,"col":6,"x":6.75,"y":5,"r":15,"rx":5.25,"ry":4,"h":2},{"label":"End","row":4,"col":7,"x":7.75,"y":6,"r":15,"rx":5.25,"ry":4},{"label":"PgDn","row":4,"col":8,"x":9.25,"y":6,"r":-15,"rx":12.75,"ry":4},{"label":"Enter","row":4,"col":9,"x":10.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":2},{"label":"Space","row":4,"col":10,"x":11.25,"y":5,"r":-15,"rx":12.75,"ry":4,"h":2},{"label":"Up","row":4,"col":11,"x":12.75,"y":4},{"label":"Down","row":4,"col":12,"x":13.75,"y":4},{"label":",","row":4,"col":13,"x":14.75,"y":4},{"label":".","row":4,"col":14,"x":15.75,"y":4.25},{"label":"mod8","row":4,"col":15,"x":16.75,"y":4.25,"w":1.25}],"keymap":{"keyboard":"foo","keymap":"","layout":"","behaviors":[{"binding":"&hm","code":"&hm","aliases":["&hm"],"identifier":"homerow_mods","bindingCells":2,"label":"HOMEROW_MODS","compatible":"zmk,behavior-hol [... cut ...] ames":[]},"message":"pls"}
```

in which I've just removed the `index` properties from each layer. The effect of this would be as if you had deleted all of the original layers through the app and created new ones, so any key bindings currently in the head commit but not in that JSON would be overwritten.

In the longer term I'll re-visit how to better handle conflicts like this.

Thanks very much for your detailed report, that request payload was extremely helpful in understanding the situation!

closed 2022-12-28 by nickcoutsos

tttaaannnggg, 2022-12-28:
(thanks so much btw!)