nickcoutsos/keymap-editor #75: Automatic Layout Generation Failed: Aurora Sweep default config

KatharinaSick, 2022-12-23:
Hello, 

I just followed the instructions at [https://zmk.dev/docs/user-setup](https://zmk.dev/docs/user-setup) to create a ZMK config for my Aurora Sweep keyboard with nice!nano v2 microcontrollers. I didn't do any customizations and simply uploaded the default config to [this repository](https://github.com/KatharinaSick/aurora-sweep-zmk-config). When opening this repo with the keymap-editor, it just displays "Automatic Layout Generation Failed". Did I do something wrong here? If I understand correctly, the `info.json` should get generated automatically?

Thanks already for your help and for creating such an amazing tool - I'm so much looking forward to playing around with it :D

Here's the summary from the config generator.
```
Preparing a user config for:
* MCU Board: nice_nano_v2
* Shield(s): splitkb_aurora_sweep_left splitkb_aurora_sweep_right
* Copy Keymap?: ✓
```

nickcoutsos, 2022-12-23:
I think you're ok -- before it attempts to fetch metadata files from the ZMK repo it'll check against a list of "known" keyboards that I update manually for the moment. It looks like I've missed some additions so I've just deployed an update. Can you give it another shot?

KatharinaSick, 2022-12-23:
It works now - thanks for your quick help! :)

nickcoutsos, 2022-12-23:
Glad to hear it!

closed 2022-12-23 by nickcoutsos