nickcoutsos/keymap-editor #76: feature request: button rotation rx,ry,r, in layout info.json

magagne, 2022-12-29:
The best will having only a rotation angle from the center of the button, the r param 0 to 90 degree, clockwise, and 0 to -90 counter clockwise. No rx neither ry. That will simplify things.

tkx nick.

nickcoutsos, 2022-12-30:
I'm trying to follow the implementation used in QMK's `info.json` which I understand is based on Keyboard Layout Editor. Removing `rx` and `ry` might simplify certain changes but may complicate others. I think it's meant to benefit rotating groups of keys like thumb clusters that are otherwise aligned together.

Whatever the reason, dealing with rotations in layouts can be annoying for sure. I've been avoiding making a layout editor but even a way to tweak the layout JSON within the app would be better than forcing people to make hundreds of commits to test things out. If I can find a good stopping point on the behavior editor work I'll try to make a helper for tweaking the layout.

caksoylar, 2023-02-15:
For what it's worth, since QMK upstream doesn't use the rotation properties they don't seem to have a specific meaning in mind as of yet: https://discord.com/channels/440868230475677696/473506116718952450/1060346223837577316

The way I decided to interpret rotation was to rotate around the center of the button as well, after translating to the top left corner of the key to the `x, y` coordinates. This seems to work OK with the few examples in the QMK repo like `afternoonlabs/breeze` which have a rotation property.

IMO KLE's rotation handling is not a good foundation because it forces you to provide `rx` `ry` everytime, otherwise it rotates around 0, 0.

I am curious about how other tools like VIA/Vial that might use (or at least use data derived from) `info.json` handles rotation but I haven't been able to investigate yet, has anyone here?

nickcoutsos, 2023-02-16:
It's been so long since I've first tried to understand this that I'm not even sure how I arrived at this implementation. As you say, QMK's tooling doesn't use rotation, and even [KLE's documentation doesn't mention it](https://github.com/ijprest/keyboard-layout-editor/wiki/Serialized-Data-Format). Recently as I tried to find more information about their rotations I found https://github.com/ijprest/keyboard-layout-editor/issues/284.

As far as I'm understanding that explanation I believe I'm using it the same way _but_ I think things are further complicated by having the origin default to the top-left corner of the key. Actually its worse because I am applying a margin between keys, so if you're trying to rotate around a specific point like the bottom left corner the offsets are a little weird. So I think defaulting to rotating around the key's center makes the most sense.

> IMO KLE's rotation handling is not great because it forces you to provide rx ry everytime, otherwise it rotates around 0, 0.

That's the global (0, 0)? Weird. Having not used KLE much I wonder the UI makes these transformations less apparently awkward. I don't want to get carried away and try to implement a half-baked version of KLE but I've seen a lot of people make a series of commits with small tweaks to try and get the editor to render the layout as they expect so there's definitely room for tooling to help with this. I've been playing around locally with a component that lets you live-edit the JSON layout:

[image]

The lines descending from the keys is pointing towards the rotation origin. I wanted to see if it expressing the Corne thumb keys as a rotation around some distant point was more appropriate than trying to iteratively position the corner of one key near the corner of the previous key and then rotate around that.

caksoylar, 2023-02-22:
Thank you for the layout editor! I also [implemented](https://github.com/caksoylar/keymap-drawer/commit/46ee88dedaed72edd90eda03ff74e4033154c4f4) `rx`/`ry` support and will use it in any custom .json files I create to ensure compatibility. I implement it like you do (translate top left to `x, y` then rotate around `rx, ry`) which I think is more reasonable than [how KLE does it](https://github.com/ijprest/keyboard-layout-editor/issues/284#issuecomment-758275435) (translate to `rx + x, ry +y` then rotate around `rx, ry`).

I think defaulting to key center if `rx, ry` aren't provided is the most convenient for users (and what I use) but using the top left corner of the key might be more correct given [the information there is in QMK docs](https://docs.qmk.fm/#/reference_info_json?id=key-dictionary-format). The issue is far from settled in the QMK spec however, I can post some example `info.json`'s from QMK repo that use these fields in conflicting ways.

nickcoutsos, 2023-02-22:
I was thinking that it might be good to use the key's center as its origin in general, but I think this is more likely to a breaking change. When you have a key wider or taller than the standard 1u size does it "grow" from the top left corner?

Here are two 1u keys over a single 2u key. In the current implementation (and I guess the "spec") with origins at the top-left corner that means both _A_ and _C_ are at `"x": 1`.

```
  ┌───────┬───────┐
  │       │       │
  │   A   │   B   │
  │       │       │
  ├───────┴───────┤
  │               │
  │       C       │
  │               │
  └───────────────┘
```

I think this makes more sense in the context of KLE and row-staggered keyboards where mixing different widths of keys doesn't require careful positioning. I'm not sure if it's enough of a problem to bother changing it as I'm not sure it's contributing anything to rotation issues.

caksoylar, 2023-02-22:
> I was thinking that it might be good to use the key's center as its origin in general, but I think this is more likely to a breaking change. When you have a key wider or taller than the standard 1u size does it "grow" from the top left corner?

At least QMK spec is defined that way, yes. `x, y` defines top left corner of the key and `w, h` values all grow from that point. I think key center might make more sense in general for that as well (and I use those in my internal representation) but given the existing QMK-format json's I am not sure it is worth breaking compatibility with it.

nickcoutsos, 2023-02-23:
Compatibility is ideal and I guess along with that would be "firmware agnosticism" but I think it may become harder to toe that line without coming up with new abstractions or treating the QMK version as canonical and adapting for ZMK keyboards. For example, QMK's `info.json` has `encoders`, but in ZMK `sensors` is used to be more generic (even though encoders are the only feature in that category).

I know this is straying from layout stuff now and becoming more about metadata in general, but would this be worth getting into in https://github.com/nickcoutsos/keymap-layout-tools and tackle the topics separately?

caksoylar, 2023-02-23:
I see your point, I admittedly hadn't considered the fields besides the physical layout.

Since ZMK officially doesn't have a concept for physical layouts as of yet (at least that I know of) sticking to the QMK format as canonical seems reasonable to me, at least for a v1 schema for metadata. I am guessing what you do here is bound to influence an eventual future design re: ZMK as well :)

Also thinking about it from a user's point of view: I can envision the most common scenario as one where users start with either a KLE export or a pre-existing QMK info.json, then tweak it in https://github.com/nickcoutsos/keymap-layout-tools if necessary. It would be easier if they didn't need a converter, at least for the QMK case.

> I know this is straying from layout stuff now and becoming more about metadata in general, but would this be worth getting into in https://github.com/nickcoutsos/keymap-layout-tools and tackle the topics separately?

Definitely!

nickcoutsos, 2023-02-23:
> I am guessing what you do here is bound to influence an eventual future design re: ZMK as well :)

That would be swell... I'm imagining the more likely scenario is that one day ZMK Studio (or a solid plan for its development) appears and I'll slowly disintegrate :)

> Also thinking about it from a user's point of view: I can envision the most common scenario as one where users start with either a KLE export or a pre-existing QMK info.json, then tweak it in [...]

You're right. I always hate hearing the old _"be conservative in what you send, be liberal in what you accept"_ thing but the fact that you and I are taking this time to better understand these existing formats suggests that it's worth having code that will handle them. That could manifest in interfaces that allow for alternate implementations or just more configurable versions of what I've already written.

I'll start some discussions in the other repo sometime tomorrow!

nickcoutsos, 2023-08-28:
So... for best compatibility I think removing `rx`/`ry` in order to default to rotations around the key center is not an option. Defaulting to the key's top-left corner for a rotation origin is not really according to spec, but I'm comfortable saying that nobody ever means to rotate around the top-left corner of the container.

I'm realizing that most difficulties in getting rotations looking correct (particularly for things like split keyboard thumb clusters since should should appear to be symmetrical) has to do with a bad decision I made early on in how I render space between keys. If a key is 50px by 50px and I want to show 5px of blank space between each key then I need to subtract that from the key dimensions and not factor it into translation/rotation. This has been fixed local dev environment for keymap-layout-tools so I'll eventually formalize that and apply it to keymap-editor's renderer as well.

Although I've always wanted to avoid keymap-layout-tools becoming a layout editor, I will eventually have to add some kind of helper for applying changes to the rotation or rotation origin of one or more keys. Being able to see changes live is helpful but it's still so tedious.

closed 2023-08-28 by nickcoutsos