nickcoutsos/keymap-editor #77: ModTap support for any keycode for hold AND any keycode for tap?

idesignstuff, 2022-12-31:
The keymap editor is a great tool for me! Thanks for making it and thanks even more for sharing it!

One little frustration I have is that &mt appears to be meant to do pretty much whatever we want on hold and whatever we want on tap.
here's a blurb from the [docs](https://zmk.dev/docs/behaviors/mod-tap): 
"Under the hood, the mod-tap is simply a [hold-tap](https://zmk.dev/docs/behaviors/hold-tap) of the ["hold-preferred" flavor](https://zmk.dev/docs/behaviors/hold-tap#flavors) with a [tapping-term-ms](https://zmk.dev/docs/behaviors/hold-tap#tapping-term-ms) of 200 that takes in two [keypresses](https://zmk.dev/docs/behaviors/key-press) as its "hold" and "tap" parameters. **This means that the mod-tap can be used to invoke any [keycode](https://zmk.dev/docs/codes), and is not limited to only activating [modifier keys](https://zmk.dev/docs/codes/modifiers) when it is held./"**

...but that's not how it's built in the editor. We get to choose from a list of modifier keys, and that's it. Would it be a difficult to open that choice up to any keycode?

idesignstuff, 2022-12-31:
For instance, I have `&mt PSCRN P` on my `P` key. It works as intended, but I can't use the editor to make it.

nickcoutsos, 2023-01-01:
Hi

I suspect that the ability to use any keycode with mod-tap is more of a happy coincidence of implementing it using the hold tap behavior. I believe the spirit of the behavior is to be used with modifiers, and the parameter definition I'm using in my implementation specifies modifiers to help with that selection. 

If you did want to create a custom behavior in your keymap using `compatible = "zmk,behavior-hold-tap";`, `bindings = <&kp>, <&kp>;` and the above mentioned properties the editor should let you assign it to a bind and select any of the known keycodes for parameters.

idesignstuff, 2023-01-01:
The docs don't explicitly describe that spirit, but they do take time to explain how any bindings can be assigned to a mod tap. The bar to create a custom hold tap is something like 20x the effort to use mod tap. 

If you don't believe it's right to support, that's okay, we can just edit the file after we're done in your editor, as I do today.

nickcoutsos, 2023-01-01:
Hold tap accepts two behaviors to trigger. Mod tap specifies `&kp` for both because there isn't one specifically for modifier codes, but the name hints that the purpose is to act as a modifier when held.

A custom hold tap behavior would like the following:

```
hp: hold_preferred {
    compatible = "zmk,behavior-hold-tap";
    label = "PREFERRED_HOLD";
    #binding-cells = <2>;
    flavor = "hold-preferred";
    tapping-term-ms = <200>;
    bindings = <&kp>, <&kp>;
};
```

and when that's found in the keymap the editor will make the behavior available, resolving the `&kp` bindings to having a single keycode parameter each. I'm working on support to create/edit custom behaviors within the application but it's in very early stages right now. Not trivial.

> we can just edit the file after we're done in your editor, as I do today.

That shouldn't be necessary either, unless you're making changes to the same binding. When the editor can't resolve a parameter's value it may show as invalid but if left untouched it should still save the original value.

idesignstuff, 2023-01-01:
Understood. Is there an equivalent to `&mt` for hold tap?

with &mt, whant I want to do looks like this:

`&mt PSCRN P`

with your suggested hp it looks like this:

hp: hold_preferred {
    compatible = "zmk,behavior-hold-tap";
    label = "PREFERRED_HOLD";
    #binding-cells = <2>;
    flavor = "hold-preferred";
    tapping-term-ms = <200>;
    bindings = <&kp PSCRN>, <&kp P>;
};

&hp

if a user wants to make many keys do a hold tap behavior, the code gets way bigger, and harder to wrangle.

nickcoutsos, 2023-01-01:
The `bindings` property in the behavior node only specifies which behaviors to use in either action, not the parameters that go along with it.

The usage in your keymap's binding would include the parameters, so altogether you'd have something like:

```
/ {
  behaviors {
    hp: hold_preferred {
      compatible = "zmk,behavior-hold-tap";
      label = "PREFERRED_HOLD";
      #binding-cells = <2>;
      flavor = "hold-preferred";
      tapping-term-ms = <200>;
      bindings = <&kp>, <&kp>;
    };
  };

  keymap {
    compatible = "zmk,keymap";

    default {
      bindings = <&hp PSCRN P ...>;
    };

    ...
  };
};
```

So its possible here to reuse `&hp` in your keymap bindings with various keycode parameters.

idesignstuff, 2023-01-01:
That explanation makes it look less unpalatable. Thank you.

idesignstuff, 2023-01-01:
I'll close this one.

closed 2023-01-01 by idesignstuff