nickcoutsos/keymap-editor #80: Different layout between layers

DMAgit, 2023-01-13:
Hey, trying to make a layout for the Ferris Sweep. The automatic generator failed, but I made one of my own. I thought I did something wrong and that it wasn't working, but I noticed that it works as intended on the non-base layers. See screenshots attached.

[image]

[image]

Also see the info.json file [here](https://github.com/DMAgit/zmk-config/blob/7fb6d31e9d105ef7e45901ddfbc3b86f7789cf3d/config/info.json)

nickcoutsos, 2023-01-13:
The layout is fine, the keymap editor doesn't support the macro you're trying to use for homerow mods.

DMAgit, 2023-01-13:
I'm sorry, could you explain how I could fix it? I'm using the base layout that comes with ZMK for the Sweep. What should I change so that it works correctly?

nickcoutsos, 2023-01-13:
Your keymap https://github.com/DMAgit/zmk-config/commit/23a5288975ee3cb80a0edaa6dcfd74fca7e7c4eb#diff-01ae956db6950cdbe4887d3c731965e52a1bd1071e198bcf82b738c1cf1919e6R43 references some macro functions to simplify the bindings for homerow mods (https://github.com/DMAgit/zmk-config/commit/23a5288975ee3cb80a0edaa6dcfd74fca7e7c4eb#diff-01ae956db6950cdbe4887d3c731965e52a1bd1071e198bcf82b738c1cf1919e6R9-R10)

To you use the editor with this keymap you would have to replace the use of that macro with the actual bindings bindings:

```diff
-HRML(A,        S,         D,         F)     &kp G          &kp H      HRMR(J,        K,         L,        SQT)
+&ht LSHFT A   &ht LALT S   &ht LCTL D   &ht LGUI F   &kp G          &kp H      &ht RGUI J   &ht RCTRL K   &ht RALT L   &ht RSHFT SQT
```

Once you do that it should be able to parse the keymap bindings correctly.

nickcoutsos, 2023-01-13:
@DMAgit I may have gotten the diff syntax wrong, you weren't meant to include the `+` at the beginning of the line.

DMAgit, 2023-01-13:
Yes, fixed that and a couple of other small things :) I think I have a better understanding of it now and its fixed. Thank you for your help 👍

closed 2023-01-13 by DMAgit

nickcoutsos, 2023-01-13:
Awesome, glad it worked out!