nickcoutsos/keymap-editor #81: Using key_repeat in hold

z5huang, 2023-01-22:


nickcoutsos, 2023-01-22:
Hmm, yeah, that would be a shortcoming in how I'm resolving the parameters for custom behaviors. The parser is doing this as generically as possible but hold-tap is a weird one since it requires 2 parameters no matter what, so I'll need to recognize that and treat the leftover parameters as "placeholders" to not trip up the UI.

In the meantime, does the keymap save correctly if you don't attempt to change that second value? Showing as missing isn't ideal, especially because it isn't missing, but generally when the editor doesn't know how to handle a value it should ignore it. As long as you don't try to alter that value it should persist when saving other changes.

I've started a discussion to share some high-level notes about the ongoing behavior editing work: https://github.com/nickcoutsos/keymap-editor/discussions/78

z5huang, 2023-01-22:
Yes it saves the keymap correctly (I made a trivial change in another key and then saved). Thanks for sharing your thoughts (and thanks for the great work in the first place!)

nickcoutsos, 2023-01-22:
Thanks for your support and kind words! I'll try and have a look at the param resolution stuff soon, there might be an easier short term fix so that it at least displays the `0` as a "raw" value.

nickcoutsos, 2023-01-22:
Just pushed a change so that placeholders are treated like raw values. It's not a proper fix by any means, but this should avoid crashing. When the behavior stuff is off my plate I can update this to automatically insert the placeholder and not try to render it in the UI.

Have a look when you have a moment and see if it works alright.

z5huang, 2023-01-22:
Thanks for the quick fix, can confirm it no longer crashes. I also tested changing the dummy `0` to `1` using the editor, and it saved the change as expected. I think this is robust enough for now. Thanks again!

I closed the issue if you don't mind

closed 2023-01-22 by z5huang