nickcoutsos/keymap-editor #82: Had an issue working with my Corne-ish Zen layout

dickiedyce, 2023-02-02:
It's a pretty minimum layout :-( but has several &mt entries. 
[corneish_zen.keymap.zip](https://github.com/nickcoutsos/keymap-editor/files/10572284/corneish_zen.keymap.zip)

nickcoutsos, 2023-02-02:
Hi, thanks for reaching out!

I can take a look at this a little later this evening, but could you provide a little detail about the nature of the issue? Did you receive any messaging in the app?

Also, would it be possible to link to a GitHub repository for your keymap? That'll make it easier to review and make suggestions than the zip file.

dickiedyce, 2023-02-03:
Hi Nick, thanks for the quick response!

https://github.com/dickiedyce/zmk-config-zen-2 is the repo in question. 

This is the error it produced:

[image]

dickiedyce, 2023-02-03:
I had been using &mt (which it also failed render against, btw), so I'd be happy to switch back to that, if that's the issue?

nickcoutsos, 2023-02-03:
Hi Richard, thanks for that!

**Short Version:**
Probably the quickest and simplest solution for you would be to grab a [pre-defined layout](https://github.com/nickcoutsos/zmk-config-corne-demo/blob/6a8527715441e0d654fcaf556aa7f1fef32bac55/config/info.json) that matches your keyboard and put it in the `config/` directory of your repo.

**Long Version:**
It doesn't look like this is related to the contents of your keymap, the error you're seeing means it's unable to generate a layout for your keyboard. Ironically (and probably frustratingly), the layout you want is visible in the background already.

If I had to guess I'd say that's the `Demo` keyboard source being rendered, and that after switching to your keymap in the `GitHub` source it's sill showing the previous keymap while attempting to fetch the new data.

When layout data isn't available in your repo the editor will try to generate one based on the keyboard's defined matrix transform. If it can't find that information in the repo it will look at the ZMK repo, relying on naming conventions to match your keymap to one of the known available keyboards. That's where things really go wrong. You have `corneish_zen.keymap` which actually is correct, except the keyboard is identified in ZMK's listing as `corneish_zen_v2`.

I hadn't encountered this before, so thank you for bringing it to my attention. The automatic layout generation is meant to be a quick and minimal effort to display a basic layout so everything is very generic (had it worked you'd see a very ortholinear rendering of the corne layout) and this is just unusual enough to trip everything up.

dickiedyce, 2023-02-04:
Thanks Nick, that's a great idea. I'll give it a go. :-)

nickcoutsos, 2023-02-07:
Hi Richard, I'm going to close this issue for now but please re-open if you're still experiencing problems.

closed 2023-02-07 by nickcoutsos