nickcoutsos/keymap-editor #83: Removing layers designation of an existing combo generates an empty list of layers, effectively disabling the combo

z5huang, 2023-02-03:
I have a few combos that were originally enabled for only a few layers. In the .keymap file these would show up as having `layers = <1,2>`, for example. I now want them to be visible to all layers, and in the GUI I would de-select the enabled layers. 

I'm not sure if that's the correct way to do it in the GUI, but it apparently results in an empty list in the generated .keymap file (i.e. something like `layers = <>`). This effectively disables the combo.

Expected output: Either remove the `layers = ...` line, or use `layers = <-1>`

nickcoutsos, 2023-02-03:
Thanks for this! I recently changed the way combo nodes are written and overlooked how empty arrays are handled. I'll try to get a fix out shortly.

z5huang, 2023-02-03:
Thanks! No worries. I just wanted to update that the `-1` layer should be in parenthesis (i.e. use `layers = <(-1)>` instead of `layers = <-1>`) or it won't compile. Will update OP

nickcoutsos, 2023-02-03:
Hm, interesting, I didn't know that would be an issue for devicetree. I haven't touched on the combo editing code in a while, you might actually run into issues in the app when it encounters `-1`... I'll update you here when it's updated to omit the `layers` property.

nickcoutsos, 2023-02-03:
@z5huang just deployed a patch, it should remove the `layers` property now when no layers are selected. I haven't tested with an existing value of `<-1>` or `<(-1)>`, if that's creating problems you might need to fix that by hand first.

Please give it a try when you can and let me know if the issue is resolved :)

z5huang, 2023-02-04:
@nickcoutsos It's working great now when removing all layers from a combo. For an existing `layers = <(-1)>` value, after selecting and then de-selecting a layer and committing the change, it _will_ generate an empty list `layers = <>`.  It's probably fine as I suspect not many people are using `layers = <(-1)>` ;)

nickcoutsos, 2023-02-04:
Thanks so much for these details!

I've gone into my code for handling integer literals and made sure to deal with negative values according to the devicetree syntax. I've also updated my combo node parser to omit `layers` if the parsed value is -1 as a short term improvement. 

At some point maybe it would be worth ignoring the default value for any property. Right now the code to handle combos is very custom, but with my work on behavior editing I have a lot of functionality built around using ZMK's schema definitions which simplifies things considerably.

I'll close this issue, but if you find I've overlooked something relating to it feel free to re-open.

closed 2023-02-04 by nickcoutsos