nickcoutsos/keymap-editor #84: [BUG] clear my keymap after commit.

kuolemax, 2023-02-09:
this is my zmk repository: [zmk-swoop](https://github.com/kuolemax/zmk-swoop)

after commit:
[image]

and my keymap file:
[image]

nickcoutsos, 2023-02-09:
Hi!

Thanks for bringing this to my attention, I'm sorry to see this causing problems for you. I'm noticing that other parts of the keymap appear to be ok (`sensor-bindings` isn't blanked out), and after looking at your layout definition it appears that you have the `"col"` and `"row"` property names swapped.

There's definitely a bug here in my code, I _think_ it would relate to the gaps in row values, but even without that you would end up with jumbled up bindings. Swap `row` and `col` (the other properties all look correct, you can leave them alone) and let me know how that goes.

kuolemax, 2023-02-10:
Thank you very much for your help, it is indeed caused by writing `col` and `row` in reverse, and now it works very well. Thanks again!

closed 2023-02-10 by kuolemax