nickcoutsos/keymap-editor #85: With more than 10 layers, the layer selection box shows an additional search box, and entering anything in there leads to UI crash

z5huang, 2023-02-17:
Once the number of layers exceeds 10 (i.e., once there is a "layer 10", which is the 11th layer), the layer selection box (the one that lets you select a layer for e.g. `&tog`) will have an additional search box above the list of layers, and entering anything into that box will lead to a crash with the error message included below. This can reproduced even with the demo repo -- just add layers til layer 10, and click on the parameter of one of the `&mo` keys on the default layer (with the demo repo, I don't even see the layer list, the UI will crash right after clicking the said parameter)
```
TypeError

e.toLowerCase is not a function

  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:205486
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:203121
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:203732
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:200277
  https://nickcoutsos.github.io/keymap-editor/main.68aec35523d370987b95.js:2:83470
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:726178
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:795302
  https://nickcoutsos.github.io/keymap-editor/main.68aec35523d370987b95.js:2:83403
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:721839
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:742751
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:779857
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:768104
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:768032
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:767893
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:764704
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:765093
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:706130
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:765542
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:675365
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:698000
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:682417
  https://nickcoutsos.github.io/keymap-editor/vendors.234ce1ceab59d99c5b64.js:2:682199
```

nickcoutsos, 2023-02-17:
Oh, that's unfortunate. I forgot about the search functionality and... I guess I've never tried that many layers. Gimme a minute, this shouldn't be too difficult to fix.

nickcoutsos, 2023-02-17:
Fixed now @z5huang, thanks for the report! Should work for any behavior parameters that expect a layer, but let me know if you run into anything else :)

closed 2023-02-17 by nickcoutsos