nickcoutsos/keymap-editor #86: 'npm run dev' fails with ERR_INVALID_URL

E5DR, 2023-02-19:
Trying to run the keymap-editor with `npm run dev` fails for me with a ERR_INVALID_URL error.

I followed the steps outlined in running locally and first cloned the repo, then created a symlink to my zmk-config into the base of the cloned repo (I also tried the example corne zmk config), after which I ran `npm install`. The error appears when trying to start the editor with `npm run dev` afterwards.
This is my first time attempting to use the editor, I might be missing something simple since nobody else seemed to have this error.

Thank you for the help!

Here the full error message:
```
[ TPMarkus ~/dev/keyboard/keymap-editor ]
$ npm run dev

> keymap-editor@1.0.0 dev
> cross-env ENABLE_DEV_SERVER=true node index.js

node:internal/url:565
  throw new ERR_INVALID_URL(input);
  ^

TypeError [ERR_INVALID_URL]: Invalid URL
    at new NodeError (node:internal/errors:399:5)
    at URL.onParseError (node:internal/url:565:9)
    at new URL (node:internal/url:645:5)
    at Object.<anonymous> (/home/markus/dev/keyboard/keymap-editor/api/index.js:12:20)
    at Module._compile (node:internal/modules/cjs/loader:1246:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1300:10)
    at Module.load (node:internal/modules/cjs/loader:1103:32)
    at Module._load (node:internal/modules/cjs/loader:942:12)
    at Module.require (node:internal/modules/cjs/loader:1127:19)
    at require (node:internal/modules/helpers:112:18) {
  input: 'undefined',
  code: 'ERR_INVALID_URL'
}

Node.js v19.6.0
```

nickcoutsos, 2023-02-19:
Looking at the location of the error it seems to be expecting the `APP_BASE_URL` environment variable to be set. Over time I went from having a simple static file server hosting the app on the same URL as the API, but that has changed. If you're starting the app server running `npm start` from within the `app/` directory, you would likely want to specify `APP_BASE_URL=http://localhost:3000`.

This should be fixed to provide a default automatically, but just to give you a warning I'm pretty heavy into development behind the scenes so the publicly available version of the code is fairly behind in terms of features. If you're fine with that I'll try to get that taken care of today or tomorrow.

If you're looking for a quicker workaround you can pass that value I mentioned, or if you're willing to dig into the code you can remove line 12 and change 15 to

```javascript
app.use(cors({ origin: '*' }))
```

E5DR, 2023-02-19:
Thank you!
While a default value will certainly be the best choice long-term, for now I am happy with the workaround.
I will do some more experimenting in the next days now that I can launch the editor. For now it appears that I have to generate a json (also, it dislikes my macros).

nickcoutsos, 2023-02-20:
Yep, this is pre-macro support. The JSON can be a little tricky to generate on your own, too. Can I ask what keyboard you're trying to edit with this? If its for a keyboard in the ZMK repo I have a selection of pre-generated layouts that I could share with you, and as I try to improve the usability of layouts I'm working on a tool to make tweaking things a little faster.

E5DR, 2023-02-21:
Well, that is how it is with the macros for now.
If you already have pre-generated layouts that would naturally be much appreciated!
My keyboard is a Sweep (34 keys split), in the ZMK repository it is the `cradio`.

nickcoutsos, 2023-02-21:
Try adding the following in `config/info.json` in your `zmk-config` directory

```json
{
  "id": "cradio",
  "name": "Cradio/Sweep",
  "layouts": {
    "LAYOUT {
      "autoGenerated": true,
      "description": "This layout was automatically generated by the Keymap Editor. It may require manual changes to display properly.",
      "layout": [
        {"row":0,"col":0,"x":0,"y":0},
        {"row":0,"col":1,"x":1,"y":0},
        {"row":0,"col":2,"x":2,"y":0},
        {"row":0,"col":3,"x":3,"y":0},
        {"row":0,"col":4,"x":4,"y":0},
        {"row":0,"col":5,"x":5,"y":0},
        {"row":0,"col":6,"x":6,"y":0},
        {"row":0,"col":7,"x":7,"y":0},
        {"row":0,"col":8,"x":8,"y":0},
        {"row":0,"col":9,"x":9,"y":0},
        {"row":1,"col":0,"x":0,"y":1},
        {"row":1,"col":1,"x":1,"y":1},
        {"row":1,"col":2,"x":2,"y":1},
        {"row":1,"col":3,"x":3,"y":1},
        {"row":1,"col":4,"x":4,"y":1},
        {"row":1,"col":5,"x":5,"y":1},
        {"row":1,"col":6,"x":6,"y":1},
        {"row":1,"col":7,"x":7,"y":1},
        {"row":1,"col":8,"x":8,"y":1},
        {"row":1,"col":9,"x":9,"y":1},
        {"row":2,"col":0,"x":0,"y":2},
        {"row":2,"col":1,"x":1,"y":2},
        {"row":2,"col":2,"x":2,"y":2},
        {"row":2,"col":3,"x":3,"y":2},
        {"row":2,"col":4,"x":4,"y":2},
        {"row":2,"col":5,"x":5,"y":2},
        {"row":2,"col":6,"x":6,"y":2},
        {"row":2,"col":7,"x":7,"y":2},
        {"row":2,"col":8,"x":8,"y":2},
        {"row":2,"col":9,"x":9,"y":2},
        {"row":3,"col":3,"x":3,"y":3},
        {"row":3,"col":4,"x":4,"y":3},
        {"row":3,"col":5,"x":5,"y":3},
        {"row":3,"col":6,"x":6,"y":3}
      ]
    }
  },
  "sensors": []
}
```

Also, just to clarify, when you're talking about macros are you talking about the ZMK behavior or preprocessor macros?

nickcoutsos, 2023-03-07:
Hi @E5DR, I finally had a chance to look at the local setup and instructions. I was missing a step to create `.env` from the provided `.env.template` which is where you'd get the default `APP_BASE_URL` value.

closed 2023-03-07 by nickcoutsos