nickcoutsos/keymap-editor #88: Feature Request: Layer names should be reflected in "label" field

NCKiser, 2023-03-10:
When changing the names of layers with the editor, the name of the layer node on the tree is changed, but the "label" field of those nodes are not changed. This means that the layer name that shows up on keyboard displays does not match the editor changes.

In this case, the OLED displays "Home" and "BT", while the editor showed the names as "Home" and "mm":
[image]
[image]

Proposal: The `label` = "__";` line for each layer node should be updated as well as the name of the layer node.

nickcoutsos, 2023-03-10:
Hi there, thank you for sharing this!

You raise a really good point. When I started this out I don't think it was common to have displays that made use of a layer's `label` property. I think such a request is what lead me to working on parsing/manipulating devicetree code directly, and the solution at the time was to manually define your labels ahead of time and the Keymap Editor would work around that property instead of deleting it.

I actually haven't gotten around to refreshing the layer editing section so I'll have to think about an approach that balances simplicity with flexibility. Probably the easiest approach would be something like:

1. Parse node name and label property (if present)
2. Render the list of keymap layers, preferring label over node name
3. Update the keymap source by re-generating the node name from the layer value
  - Devicetree node names have some restrictions on valid characters
  - The actual node name itself is less meaningful than a label since nothing needs to reference it

Even so, I feel like I'd prefer to avoid doing unexpected things too much. The other approach would be to adapt something similar to the combo/macro/behavior editors and pop up a modal where the label can be edited separately.

nickcoutsos, 2023-03-16:
Hi @NCKiser, I've been moving some things around in the layer editing components and I've added a configuration dialog (accessible via a new three-dot menu next to the _Key Bindings_ heading) where you can view/change the `label` text for the active layer.

closed 2023-03-30 by nickcoutsos