nickcoutsos/keymap-editor #89: Error

pujvz, 2023-03-23:
```
TypeError

o is undefined

  webpack://app/src/ValuePicker/SearchFilter.js:18:21
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:289:385
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:279:391
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:279:322
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:270:87
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:272:299
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:127:104
  webpack://app/node_modules/react-dom/cjs/react-dom.production.min.js:266:268
```

pujvz, 2023-03-23:
It also looks like the editor is automatically renaming macros, which results in devicetree errors while building.

This is an example from the original keymap:
```
    macros {
        **bt_0**: bt_profile_macro_0 {
            label = "BT_0";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings
                = <&out OUT_BLE>,
                  <&bt BT_SEL 0>;
        };

and this is what changes after the keymap is loaded in the web app editor:
```
    macros {
        **bt_profile_macro_0**: bt_profile_macro_0 {
            label = "BT_0";
            compatible = "zmk,behavior-macro";
            #binding-cells = <0>;
            bindings = <&out OUT_BLE &bt BT_SEL 0>;
        };

nickcoutsos, 2023-03-24:
Hey thanks for raising this! Can you elaborate a little on the initial error? What sort of behavior were you selecting a value for?

I'll have a look at the renaming, that's definitely not good!

pujvz, 2023-03-24:
Sorry that was confusing. I probably should have created a separate issue for the second comment.

The behavior that caused the first error was clicking on a hold tap key with a &mo, macro. This is the mapping for this key:
```
    magic: magic_hold_tap {
        compatible = "zmk,behavior-hold-tap";
        label = "MAGIC_HOLD_TAP";
        #binding-cells = <2>;
        flavor = "tap-preferred";
        tapping-term-ms = <200>;
        bindings = <&mo>, <&rgb_ug_status_macro>;
    };
```
[image]
Clicking either button indicated by red arrow causes the error.

nickcoutsos, 2023-03-24:
Can you click the copy button in that dialog? I think there are more lines not being shown.

I'll have a go at reproducing that locally based on your sample behavior.

nickcoutsos, 2023-03-24:
Ok, trying to bind parameters to a hold-tap that uses a macro as part of its own binding reliably causes the crash. I don't treat macros like other behaviors and this likely causes a problem during parameter resolution. It shouldn't be a difficult fix but I'd better leave it for tomorrow and get some sleep.

Thanks again for catching this, I'll reach out when it's resolved 👍

pujvz, 2023-03-24:
Sure, I pasted the full error below. I appreciate the quick replies, but no need to hurry! Most of the things I need are already working. 

BTW I was renaming the macros in a text editor prior to your suggestion. But doing the renaming in the web app macro tab did the trick :).

```
TypeError

o is undefined

  webpack:///src/ValuePicker/SearchFilter.js:18:21
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:289:385
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:391
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:322
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:270:87
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:272:299
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:127:104
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:266:268
```

nickcoutsos, 2023-03-24:
Ok, so there are a few factors at play:

1. Unrecognized behaviors in a hold-tap should resolve to a "raw" parameter, but the representation of this was inconsistent so it wasn't choosing the correct value picker component. **This is fixed now.**
2. When resolving parameters for a behavior, if any behaviors referenced in its `bindings` are unrecognized the resolution is abandoned and all parameters are treated as a "raw" value type, even if some could be resolved. **This is no longer the case** but I'm not sure I'm happy with the approach.
3. Macros aren't considered during behavior resolution, mostly just due to how all of the data is structured. This _can_ be fixed but I don't think I will bother doing anything here until I've consolidated how macros and behaviors are managed in the backend.

At a high level, the current state of this is that you can now assign macros to hold-tap (and other) behaviors and then bind those behaviors in your keymap. It won't correctly see that the macro doesn't require a parameter so instead of injecting a placeholder and hiding that parameter it will expect you to provide some value yourself.

pujvz, 2023-03-25:
Thanks for the update! I tested it and the web app no longer crashes when I try to change a macro key value.

pujvz, 2023-03-26:
I have been playing around with the macro tab, and I ave spotted some quirks. 

- Currently it's not possible to delete a macro. Clicking the trashcan icon has no effect.
- If a new macro is created in the web app and I try adding key binding or control binding, I am unable to delete either.
- If a new macro is created and I try adding a string sequence, I am only able to delete part of the key bindings, but not all of them.
- Elements in already defined macros can only be deleted some of the time. In my current keymap, I'm only able to delete elements in the first defined macro.

nickcoutsos, 2023-03-26:
Separate issue would be great, thank you! They seem to be related and I wouldn't be surprised if they all share a similar root cause. I'll close this one and take a peek at the macro handling events.

closed 2023-03-26 by nickcoutsos

nickcoutsos, 2023-03-26:
@pujvz as suspected there were some stupid mistakes in the delete handler that I've corrected. I didn't test too thoroughly, but based on the description and the effects of the mistake it would make sense that it would cause a lot of confusing behavior. Feel free to confirm based on your experience of the issue, but it looks it may not need any more effort.

pujvz, 2023-03-26:
I did a quick test and everything seems to be delete-able now thanks! I will definitely open separate issues in the future. Sorry about that!