nickcoutsos/keymap-editor #90: Feature request: remove Confirm button or add a setting to auto-confirm on keycode selection

pujvz, 2023-03-24:
I think this feature would make editing keymaps a bit faster and easier.
**Feature** - upon selecting the desired keycode, the keycode is applied to the keymap without needing to click Confirm.

[image]
For example in this screenshot, immediately after clicking `RIGHT_PARENTHESIS`, the prompt is automatically closed and the change is applied without ever clicking Confirm.

nickcoutsos, 2023-03-24:
You're totally right, it's something that's been annoying me as well. I need to balance convenience with flexibility as this particular dialog differs from most of the typical pickers in that there are multiple elements that may require interaction before the user is done and I can't dictate a particular order of events.

As a first pass it would be reasonable to enable `Enter` to at least focus the `Confirm` button, but maybe some experimentation could spark more ideas.

Separately, if you're trying to set a larger number of bindings (or an entire layer), have you seen the _Quick `&kp` Assignment_ feature? You can access it from the three-dot menu next to the _Key Bindings_ heading and it just iterates through every key, letting you use your keyboard to assign keycode bindings without bringing up the picker.

pujvz, 2023-03-24:
> have you seen the Quick &kp Assignment feature?

:exploding_head: That's pretty neat! I had not even noticed the three-dot menu. I will definitely be using this feature.

nickcoutsos, 2023-03-25:
It's quite new -- I only added it a week or so back, but as I add ways to streamline things I also need make sure I can clue users in to the new options. When I added drag selection for key position widgets I included a component that tracks a number of interactions before suggesting an easier way. I could possibly do something similar here.

danielo515, 2023-03-31:
> It's quite new -- I only added it a week or so back, but as I add ways to streamline things I also need make sure I can clue users in to the new options. When I added drag selection for key position [...]

That feature is quite neat, I was very happy when I discovered it. However, I think that the process of setting new key from scratch can be easier. Ideally, it should work as a wizard if more than one thing needs to be configured. As soon as you select kp then move to the next input element, which is the key, and then allow you to just hit enter or whatever to be able to save it

nickcoutsos, 2023-10-03:
I think I've looked at this a few times now and it only now just clicked. (I had a lot of other things on the go, sorry to let the conversation hang like that)

What you're describing is something I'd thought about in the past, but with recent refactors around bind editing and moving these actionable elements into a dialog I think that kind of guided flow is much more achievable! I'm going to keep this in the back of my head for a bit but realistically it shouldn't be a huge effort to get there.