nickcoutsos/keymap-editor #94: Behaviours get relabeled.

danielo515, 2023-03-31:
Hello, and thank you for this awesome project.
Want to port my QMK combos and stuff to ZMK, and  having this easy to digest way of doing it is amazing.

I want to use to for my glove80, and I am having a little problem: behaviours are modified so their name matches their label. I understand this is a convenience way of preventing errors in the code, but  it is making my build to fail, because those behaviours are referenced in the keymap, and I am not sure if the other name (sorry, don't know what is the label and what is  the other thing) can be safely renamed.

Here is an example diff of what I mean:

https://github.com/danielo515/glove80-zmk-config/pull/1/files#diff-851c976491c4ffabdf14ba798d167e623eaf33acc918702c109b1549c43cc504R41-R43

[image]

nickcoutsos, 2023-03-31:
This is specific to my (older) macro implementation and not behaviors in general. At the moment it doesn't give you the flexibility to control the name (the node name, `rgb_ug_status_macro_0` in your case) and the "bind" (in devicetree terms this is a "label" but I sometimes avoid calling it that so as not to confuse it with the `label` property). 

Conventially a lot of people will use the same value for all three places which is part of what led to this implementation. I'll eventually update the macro editor to give you more control here but in the meantime you would need to match the binding up with the name to avoid the editor breaking things.

danielo515, 2023-04-03:
Sorry, I am completely new to ZMK and I just pasted whatever the glove80 layout editor spits out. I guess that editor also has it's own internal nuances, and it is not probably worrying much about the quality or normality of what it outputs as soon as it compiles.

I just renamed all the macros and behaviours and it just works fine, so I guess this is not that big of a deal.
Regars

danielo515, 2023-04-04:
Hello @nickcoutsos did you introduced that feature already? I saw in the config for the macros that you can define the node and the label, along with the name, that's great, thanks.
The bad part is that I discovered it because, although all the new macros were getting the right label, the name was being pick as new_macro, resulting in several macro definitions with the same name. I want to emphasise that I put the name in the usual place, but it is not picked as default value for the other fields, resulting in duplicated macro names.

nickcoutsos, 2023-04-04:
Click the wrench icon and you'll see new fields.

danielo515, 2023-04-05:
> Click the wrench icon and you'll see new fields.

Yes, and that is what I did. However, the problem is that this makes it impossible to build any new macro with the default flow, because it will be labelled new_macro unless you click the wrench and edit the other fields.
Also, in my case, after doing that, for some reason they were not being applied, all my new macros ended being named new_macro.

nickcoutsos, 2023-04-05:
I'll be changing the flow to display the modal when creating new macros.(

I'm not sure what you mean in the other part. You changed the fields in the modal dialog but they didn't persist? I'll look into it but I haven't been able to reproduce it so far. Then next time you're in the app could you try refreshing with `Cmd+Shift+R`? (assuming you're on a Mac, might be `Ctrl+Shift+R` or `Ctrl+Shift+F5` on Windows)

nickcoutsos, 2023-04-13:
Since I can't reproduce the renaming problem I'm going to close the issue, but if you have more details feel free to re-open.

closed 2023-04-13 by nickcoutsos