nickcoutsos/keymap-editor #96: Unexpected Error upon opening the web app

pujvz, 2023-04-04:
When opening the web app I am getting this error. I pasted the complete output below:

"Unexpected error

Something unexpected happened and the keymap editor app cannot recover. Please consider taking a moment to open a new issue with the details below to help me investigate and improve the app."

```
TypeError

n is undefined

  webpack:///src/Keyboard/Warnings/sanity-checks.js:29:30
  webpack:///src/Keyboard/Warnings/sanity-checks.js:29:18
  webpack:///src/Keyboard/Warnings/sanity-checks.js:53:18
  webpack:///src/Keyboard/Warnings/sanity-checks.js:12:15
  webpack:///src/Keyboard/Keyboard.js:62:49
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:182:188
  webpack:///node_modules/react/cjs/react.production.min.js:25:207
  webpack:///src/Keyboard/Keyboard.js:62:23
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:166:136
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:289:385
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:391
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:322
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:270:87
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:272:299
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:127:104
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:266:268
```

nickcoutsos, 2023-04-04:
I'm having trouble reproducing this, do you have a sample keymap you could share? It looks like it's macro related, so if you could even just the macro nodes of your keymap that could help.

pujvz, 2023-04-04:
Weird I'm getting the error without even logging in in a fresh browser profile.

nickcoutsos, 2023-04-04:
Thank you, that helps quite a bit actually!

nickcoutsos, 2023-04-04:
I just pushed a fix that should take care of it, as long as the problem is what I suspect. If not, I'm gonna need more detail :)

pujvz, 2023-04-04:
Awesome! It's fixed now thanks :smile: .

closed 2023-04-04 by pujvz