nickcoutsos/keymap-editor #98: Can't save keymaps anymore

afebles, 2023-04-05:
```
TypeError

Cannot read properties of undefined (reading 'match')

  https://nickcoutsos.github.io/keymap-editor/main.7ae8aca32d7b92d49402.js:2:204364
  https://nickcoutsos.github.io/keymap-editor/main.7ae8aca32d7b92d49402.js:2:204337
  https://nickcoutsos.github.io/keymap-editor/main.7ae8aca32d7b92d49402.js:2:204267
  https://nickcoutsos.github.io/keymap-editor/main.7ae8aca32d7b92d49402.js:2:213514
  https://nickcoutsos.github.io/keymap-editor/main.7ae8aca32d7b92d49402.js:2:213613
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:177:118
  webpack:///node_modules/react/cjs/react.production.min.js:25:207
  https://nickcoutsos.github.io/keymap-editor/main.7ae8aca32d7b92d49402.js:2:213107
```

nickcoutsos, 2023-04-05:
It looks like you've been able to save since making this issue, can you tell me a little more about what steps you had taken before arriving at this error?

afebles, 2023-04-05:
Sorry, I'm new to this and I love your tool. 
I've been able to commit my changes but my builds keep on failing. Don't think that has to do with your tool but here's the failing code:

##[debug]Evaluating condition for step: 'West Build (${{ env.display_name }})'
##[debug]Evaluating: format('West Build ({0})', env.display_name)
##[debug]Evaluating format:
##[debug]..Evaluating String:
##[debug]..=> 'West Build ({0})'
##[debug]..Evaluating Index:
##[debug]....Evaluating env:
##[debug]....=> Object
##[debug]....Evaluating String:
##[debug]....=> 'display_name'
##[debug]..=> 'corneish_zen_v[2](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:2)_left'
##[debug]=> 'West Build (corneish_zen_v2_left)'
##[debug]Result: 'West Build (corneish_zen_v2_left)'
##[debug]Set step '__run_5' display name to: 'West Build (corneish_zen_v2_left)'
##[debug]Evaluating: success()
##[debug]Evaluating success:
##[debug]=> true
##[debug]Result: true
##[debug]Starting: West Build (corneish_zen_v2_left)
##[debug]Loading inputs
##[debug]Evaluating: format('west build -s zmk/app -b {0} -- -DZMK_CONFIG=${{GITHUB_WORKSPACE}}/{1} {2} {[3](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:3)}', matrix.board, inputs.config_path, env.extra_cmake_args, matrix.cmake-args)
##[debug]Evaluating format:
##[debug]..Evaluating String:
##[debug]..=> 'west build -s zmk/app -b {0} -- -DZMK_CONFIG=${{GITHUB_WORKSPACE}}/{1} {2} {3}'
##[debug]..Evaluating Index:
##[debug]....Evaluating matrix:
##[debug]....=> Object
##[debug]....Evaluating String:
##[debug]....=> 'board'
##[debug]..=> 'corneish_zen_v2_left'
##[debug]..Evaluating Index:
##[debug]....Evaluating inputs:
##[debug]....=> Object
##[debug]....Evaluating String:
##[debug]....=> 'config_path'
##[debug]..=> 'config'
##[debug]..Evaluating Index:
##[debug]....Evaluating env:
##[debug]....=> Object
##[debug]....Evaluating String:
##[debug]....=> 'extra_cmake_args'
##[debug]..=> ''
##[debug]..Evaluating Index:
##[debug]....Evaluating matrix:
##[debug]....=> Object
##[debug]....Evaluating String:
##[debug]....=> 'cmake-args'
##[debug]..=> null
##[debug]=> 'west build -s zmk/app -b corneish_zen_v2_left -- -DZMK_CONFIG=${GITHUB_WORKSPACE}/config  '
##[debug]Result: 'west build -s zmk/app -b corneish_zen_v2_left -- -DZMK_CONFIG=${GITHUB_WORKSPACE}/config  '
##[debug]Loading env
Run west build -s zmk/app -b corneish_zen_v2_left -- -DZMK_CONFIG=${GITHUB_WORKSPACE}/config  
##[debug]sh -x /__w/_temp/d381[4](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:4)a[5](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:5)e-cf8[6](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:6)-415f-b0f5-bc1[7](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:7)[8](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:8)4a34e8b.sh
+ west build -s zmk/app -b corneish_zen_v2_left -- -DZMK_CONFIG=/__w/zmk-config-zen-2/zmk-config-zen-2/config
-- west build: generating a build system
Including boilerplate (Zephyr base): /__w/zmk-config-zen-2/zmk-config-zen-2/zephyr/cmake/app/boilerplate.cmake
-- Application: /__w/zmk-config-zen-2/zmk-config-zen-2/zmk/app
-- ZMK Config directory: /__w/zmk-config-zen-2/zmk-config-zen-2/config
-- ZMK Config Kconfig: /__w/zmk-config-zen-2/zmk-config-zen-2/config/corneish_zen.conf
-- Using keymap file: /__w/zmk-config-zen-2/zmk-config-zen-2/config/corneish_zen.keymap
-- Using keymap file: /__w/zmk-config-zen-2/zmk-config-zen-2/config/corneish_zen.keymap
-- Zephyr version: 3.0.0 (/__w/zmk-config-zen-2/zmk-config-zen-2/zephyr)
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.[10](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:10)") found components: Interpreter 
-- Found west (found suitable version "0.[14](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:14).0", minimum required is "0.7.1")
-- Board: corneish_zen_v2_left
-- Cache files will be written to: /__w/zmk-config-zen-2/zmk-config-zen-2/zephyr/.cache
-- ZEPHYR_TOOLCHAIN_VARIANT not set, trying to locate Zephyr SDK
-- Found host-tools: zephyr 0.[15](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:15).2 (/opt/zephyr-sdk-0.15.2)
-- Found dtc: /opt/zephyr-sdk-0.15.2/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6")
-- Found toolchain: zephyr 0.15.2 (/opt/zephyr-sdk-0.15.2)
-- Found BOARD.dts: /__w/zmk-config-zen-2/zmk-config-zen-2/zmk/app/boards/arm/corneish_zen/corneish_zen_v2_left.dts
-- Found devicetree overlay: /__w/zmk-config-zen-2/zmk-config-zen-2/config/corneish_zen.keymap
devicetree error: expected property 'bindings' in /macros/new_macro in /__w/zmk-config-zen-2/zmk-config-zen-2/zephyr/misc/empty_file.c to be assigned with 'bindings = < &foo ... &bar 1 ... &baz 2 3 >' (a mix of phandles and numbers), not 'bindings;'
CMake Error at /__w/zmk-config-zen-2/zmk-config-zen-2/zephyr/cmake/dts.cmake:[20](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:20)5 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
[... cut ...]
  /__w/zmk-config-zen-2/zmk-config-zen-2/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:[35](https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:35) (include_boilerplate)
  CMakeLists.txt:15 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake -DWEST_PYTHON=/usr/bin/python3 -B/__w/zmk-config-zen-2/zmk-config-zen-2/build -S/__w/zmk-config-zen-2/zmk-config-zen-2/zmk/app -GNinja -DBOARD=corneish_zen_v2_left -DZMK_CONFIG=/__w/zmk-config-zen-2/zmk-config-zen-2/config
Error: Process completed with exit code 1.
##[debug]Finishing: West Build (corneish_zen_v2_left)

nickcoutsos, 2023-04-05:
FYI you can link directly to the failing build (and even a particular line) which helps a fair bit for people trying to debug. Here's the error in your first failing build: https://github.com/afebles/zmk-config-zen-2/actions/runs/4616925120/jobs/8162600127#step:9:81

It's complaining about a missing `bindings` array and in your [keymap](https://github.com/afebles/zmk-config-zen-2/blob/459b00a24eb1dc3b10d19752dda2b9b5eca2e341/config/corneish_zen.keymap#L38) you can see that there's a new macro without any bindings included. You can either add some bindings there or delete the macro if you don't want to use it.

At some point I'll have time to add a little more validation to warn you about saving with missing values.

afebles, 2023-04-05:
Ahhhh genius! That makes sense. Thank you for the guidance and tips

closed 2023-04-05 by nickcoutsos