nickcoutsos/keymap-editor #102: Error when loading Webapp, but works again after reloading page. Happened twice today

Vieavona, 2023-04-21:
```
ChunkLoadError

Loading chunk 832 failed.
(error: https://nickcoutsos.github.io/keymap-editor/832.ab56e6006c60b5bf5639.js)

  https://nickcoutsos.github.io/keymap-editor/runtime.557b86a02fc3cf497e45.js:1:3705
  https://nickcoutsos.github.io/keymap-editor/runtime.557b86a02fc3cf497e45.js:1:1412
  [native code]:undefined:undefined
  https://nickcoutsos.github.io/keymap-editor/runtime.557b86a02fc3cf497e45.js:1:1383
  https://nickcoutsos.github.io/keymap-editor/main.b6b64e48edce2dabdf00.js:2:117061
  webpack:///node_modules/react/cjs/react.production.min.js:18:160
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:290:394
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:391
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:322
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:279:179
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:267:211
  [native code]:undefined:undefined
  https://nickcoutsos.github.io/keymap-editor/vendors.763936cf2033c3d68135.js:2:439196
  https://nickcoutsos.github.io/keymap-editor/vendors.763936cf2033c3d68135.js:2:439728
```

nickcoutsos, 2023-04-21:
Hmm. I've never encountered this sort of error before but the fact that its related to loading a static file and that refreshing made it go away points toward a networking issue. I'll look into it a little more but if you can share anymore detail that would be great.

nickcoutsos, 2023-04-22:
Looking again I think the issue was related to the browser caching deprecated files. I'll have a look at my deploy process to try and mitigate problems like that.

Vieavona, 2023-05-07:
Sorry for the very late response, but yes, I think that it had to do something to do with the caching as well. It doesn't really bother me and I've not experienced it since, it just happened twice and I thought "Yeah, why not report it". In my perspective, this issue can even be closed, as I probably wouldn't want to work on it since it's not really a hindrance in using the app.

nickcoutsos, 2023-05-07:
Cool, thank you! I'm no longer deleting the old deployed versions of the app code so that should mostly mitigate this. At some point I'll have to consider a "cleanup" to remove the oldest files and probably also make sure `index.html` can only be cached for an hour or so.

closed 2023-05-07 by nickcoutsos