nickcoutsos/keymap-editor #103: Using #defined macro-name as key behavior breaks parsing of the keymap file

eegil, 2023-04-23:
having the following practical example: 
`
#define AS(keycode) &as LS(keycode) keycode     // Autoshift Macro
/ {
    behaviors {
        as: auto_shift {
            .....
        };
    };
  keymap {
        compatible = "zmk,keymap";
        default_layer {
            bindings = <
                AS(Q) AS(W) AS(E) AS(R) AS(T) AS(Y) // Autoshift applied for QWERTY keys
            >;
        };
    };
};
`
Firmware build goes fine and key behavior works as expected. 

Issue: When trying to open the keymap with the configurator the following message is displayed: **Could not parse keymap**

nickcoutsos, 2023-04-23:
Using C macros to replace behaviors isn't supported because in addition to the extra complication to parse them it isn't feasible to attempt to reproduce it when saving changes. In this case its further complicated because the parser attempts to group bind parameters with their behavior but the first element in the array isn't a behavior, so you aren't getting the intended error message:

> It looks like some bindings in your keymap are using custom pre-processor
> macros which isn't supported by the Keymap Editor. These macros make
> it difficult both to resolve bindings to behaviors and parameters, as
> well as to generate modified keymap code that is consistent with your
> original intentions.

> Please either continue editing your keymap by hand with these helpers
> or if you would like to use this editor, rewrite the keymap to use
> ZMK's syntax directly.

closed 2023-04-23 by nickcoutsos

autoferrit, 2023-05-04:
I am having the same issue and found this. is there a better way to write macros like this auto_shift macro in ZMK language?

nickcoutsos, 2023-05-04:
Not really?

I think ideally there would be a more suitable behavior, maybe just a built-in auto shift so that you don't need to repeat the keycode argument or something _slightly_ more generic that lets you specify an alternate modifier. Without that your options are:

1. Just write out the binding in long form: `&as LS(A) A` which isn't fun to read, unfortunately, but that's less of an issue if you want to prioritize reading _and_ writing keymaps with this tool.
2. Use the macro which simplifies the code so that you don't need this editor.

While I _could_ parse the macro into editable bindings, it wouldn't be worth the trouble because doing the opposite (choosing to use the preprocessor macro when encountering a key binding where it could conceivably be used) would make things exponentially more complex.

Going by your question on the ZMK discord server I want to make sure I'm not overlooking a detail. The auto shift described here is just a hold-tap where the "hold" behavior is the same keycode with a `MOD_LSFT` applied, and not the "sentence based" auto-capitalization you were asking about.

autoferrit, 2023-05-05:
Yes i agree that the senstence based capitalization is something totally different. This was something i have been playing with using for simply holding some keys, namely symbols, to get their shifted versions. like `\,./'` etc. But im thinking that for those few keys a hold tap might be better anyways.