nickcoutsos/keymap-editor #104: Macros key positions are incorrect for the Adv360

RATIU5, 2023-05-03:
Added some macros to my adv360 layout, some did not work. Example:

[image]

results in

```c
backslash {
    bindings = <&kp BACKSLASH>;
    key-positions = <31 51>;
    timeout-ms = <50>;
    layers = <0 2 3>;
};
```

but according to the keymap taken by urob's config, the Adv360 has this map:

```c
/*               ADVANTAGE 360 KEY MATRIX
 ╭────────────────────────────┬────────────────────────────╮
 │  0   1   2   3   4   5   6 │  7   8   9  10  11  12  13 │
 │ 14  15  16  17  18  19  20 │ 21  22  23  24  25  26  27 │
 │ 28  29  30  31  32  33  34 │ 35  36  37  38  39  40  41 │
 │ 42  43  44  45  46  47 ╭───┴───╮ 48  49  50  51  52  53 │
 │ 54  55  56  57  58╭────╯       ╰────╮59  60  61  62  63 │
 ╰───────────────────┼────────┬────────┼───────────────────╯
                 ╭───╯ 64  65 │ 66  67 ╰───╮
                 │ 68  69  70 │ 71  72  73 │
                 ╰───────╮ 74 │ 75 ╭───────╯
                         ╰────┴────╯ */
```

According to the combo, it's selecting two keys on opposite sides of the board. I haven't looked through the codebase for a solution but I at least wanted to make the bug known. Great work on this tool btw!

nickcoutsos, 2023-05-03:
So, uh, the Advantage 360 layout has been a bit of a mess but:

1. Are you sure that you're using urob's matrix transform? Somewhere in your devicetree with a `chosen` node that references it and although I haven't examined your repo too closely I would expect that mismatched layout and matrix transform would fuck up... everything.
2. The pre-defined layout wouldn't be set up to match that matrix transform (a simple bit crucial example is the omission of foot pedals). You could try adjusting it or creating a new one with the help of https://nickcoutsos.github.io/keymap-layout-tools/ but it's not trivial so it's worth validating your other options first.

nickcoutsos, 2023-05-04:
Any news on this @RATIU5?

RATIU5, 2023-05-04:
My config was a direct fork from [Kinesis' repo](https://github.com/KinesisCorporation/Adv360-Pro-ZMK). The only thing I did was open the repo inside your layout tool and configured away. I referenced urob's matrix after I realized the combos did not work correctly.
I don't believe I used his matrix transforms in this project, I was however using them in a different repo.

nickcoutsos, 2023-05-05:
Based on the current JSON in the repo you would use these key positions:

[image]

Note that some positions are skipped -- my understanding is that those positions in the transformation matrix are intended for foot pedals but aren't displayed in the editor.

closed 2023-05-05 by nickcoutsos