nickcoutsos/keymap-editor #108: Unexpected error

Poet-Li, 2023-05-14:
I just tried the Clipboard and then I got the error message:

Something unexpected happened and the keymap editor app cannot recover. Please consider taking a moment to open a new issue with the details below to help me investigate and improve the app.

```
TypeError

Cannot convert undefined or null to object

  webpack:///src/Sources/Clipboard/Picker.js:20:33
  webpack:///src/Sources/Clipboard/Picker.js:38:6
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:242:331
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:285:110
  webpack:///node_modules/react-dom/cjs/react-dom.production.min.js:281:390
  webpack:///node_modules/scheduler/cjs/scheduler.production.min.js:13:202
  webpack:///node_modules/scheduler/cjs/scheduler.production.min.js:14:127
```

My clipboard contents are as follow:

json:

```
[
  [
    "Q",
    "W",
    "E",
    "R",
    "T",
    "Y",
    "U",
    "I",
    "O",
    "P"
  ],
  [
    "A",
    "S",
    "D",
    "F",
    "G",
    "H",
    "J",
    "K",
    "L",
    ";"
  ],
  [
    "Z",
    "X",
    "C",
    "V",
    "B",
    "N",
    "M",
    ",",
    ".",
    "/"
  ],
  [
    {
      "x": 2.5,
      "a": 7
    },
    "",
    {
      "w": 1.5
    },
    "",
    {
      "w": 1.5
    },
    "",
    ""
  ]
]
```

and the .keymap:

```
/*
* Copyright (c) 2021 The ZMK Contributors
*
* SPDX-License-Identifier: MIT
*/

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#include <dt-bindings/zmk/bt.h>

#define MAIN 0
#define SYM 1
#define NAV 2
#define BT 3

&mt {
    flavor = "tap-preferred";
    tapping_term_ms = <140>;
};

/ {
    combos {
        compatible = "zmk,combos";

        combo_esc {
            timeout-ms = <100>;
            key-positions = <21 22>;
            bindings = <&kp ESC>;
        };

        combo_tab {
            timeout-ms = <100>;
            key-positions = <22 23>;
            bindings = <&kp TAB>;
        };

        combo_minus {
            timeout-ms = <100>;
            key-positions = <26 27>;
            bindings = <&kp MINUS>;
        };

        combo_underscore {
            timeout-ms = <100>;
            key-positions = <26 28>;
            bindings = <&kp UNDERSCORE>;
        };

        combo_colon {
            timeout-ms = <100>;
            key-positions = <7 8>;
            bindings = <&kp COLON>;
        };

        combo_semicolon {
            timeout-ms = <100>;
            key-positions = <6 8>;
            bindings = <&kp SEMICOLON>;
        };

        combo_backslash {
            timeout-ms = <100>;
            key-positions = <27 28>;
            bindings = <&kp BSLH>;
        };

        combo_grave {
            timeout-ms = <100>;
            key-positions = <8 9>;
            bindings = <&kp GRAVE>;
        };
    };

    keymap {
        compatible = "zmk,keymap";

        MAIN_layer {
            bindings = <
                &kp Q       &kp W      &kp E       &kp R        &kp T            &kp Y     &kp U      &kp I           &kp O        &kp P
                &kp A       &kp S      &kp D       &lt SYM F    &kp G            &kp H     &lt SYM J  &kp K           &kp L        &kp SQT        
                &mt LSHFT Z &mt LALT X &mt LCTRL C &mt LGUI V   &kp B            &kp N     &mt RGUI M &mt RCTRL COMMA &mt RALT DOT &mt RSHFT FSLH
                                                   &lt BT ENTER &lt NAV SPACE    &sk RSHFT &kp BSPC
            >;
        };

    };
};
```

It crashed because I entered the part before the device tree in the .keymap file?

nickcoutsos, 2023-05-14:
No, but your layout is wrong. May I ask how you came up with it? It looks like KLE's format which isn't supported. If your keyboard is supported by QMK you can try looking for its accompanying `info.json` in the QMK repo. If you need to make tweaks:

- [Keymap Layout Helper](https://nickcoutsos.github.io/keymap-layout-tools/) allows you to visualize layouts and test changes on the fly
- [Wiki: Defining keyboard layouts](https://github.com/nickcoutsos/keymap-editor/wiki/Defining-keyboard-layouts) is my description of the layout format and how I deviate from QMK's spec to support formatting the bindings array in devicetree.

Poet-Li, 2023-05-14:
Sorry for my reading comprehension mistake, the wrong format layout is the json generated by the KLE website(http://www.keyboard-layout-editor.com/). 
It works well. Thank you!!

closed 2023-05-14 by Poet-Li

nickcoutsos, 2023-05-14:
That's a legitimate issue in the app; I have basically no error-checking in this component right now and the data you enter is persisted in local storage so depending on the problem you could end up stuck unfortunately.

I have some work on the go for the clipboard stuff but in the meantime you can run something like `localStorage.clear()` in your dev console and refresh the page.