nickcoutsos/keymap-editor #110: zmk, behaviour hold-tap

red-four, 2023-06-02:
Attempting to use a custom behaviour for hold-tap; not able to save. 

[image]
[image]

(If there is a better way to get this option, I am here for it).

nickcoutsos, 2023-06-02:
Hi, thanks for the report! It looks like you're doing things correctly there as all of the required fields are filled in. I'm trying to reproduce this effect myself with the same parameters and it only happens if I haven't checked the _Enable Beta_ checkbox. Could you confirm if that's the case here?

red-four, 2023-06-02:
I had the enable beta box clicked when I was making this.

nickcoutsos, 2023-06-02:
Can you check the browser console for errors? If you're clicking Okay and nothing happens then that would suggest an error that isn't being caught.

nickcoutsos, 2023-06-02:
Wait, I'm seeing in your other screenshot that the behavior exists because it's already bound to some keys. By save you mean commit? I'm seeing error logs for `red-four/reviung41-v3`, is that the repo in question? It looks like you've _removed_ it from your installation which means the app no longer has write permissions.

red-four, 2023-06-02:
... ohhh. because I deleted and then remade it, but didn't give the new one permissions. That would do it.

nickcoutsos, 2023-06-03:
Awesome, glad to hear that!

closed 2023-06-03 by nickcoutsos