nickcoutsos/keymap-editor #111: Unable to use the 'Enter' key to confirm a key code selectio

RobertWCarey, 2023-06-07:


nickcoutsos, 2023-06-07:
Thanks!

I can reproduce the issue and have pinpointed the source (a key event handler is inappropriately consuming the _Enter_ keypress to select the highlighted key from the list), but in the middle of some work that touches on this and won't be able to deploy a fix as quickly as I'd like.

As a temporary workaround you can also press _Space_ to trigger the focused button element, but I'll keep this issue open and update you once it's properly resolved.

nickcoutsos, 2023-06-08:
Alright, well, my other task ended up being too complex to tackle at once so I wrapped it up and put in a quick fix for this issue. Thank you again for sharing!

closed 2023-06-08 by nickcoutsos