nickcoutsos/keymap-editor #112: Making a commit: firmware button behavior

WainingForests, 2023-06-10:
When making a commit, the firmware "blue button" stays on the old build action instead of taking you to the newest action (even if it's not complete). Is it possible to have it take you to the most recent action; even if it's still building?

WainingForests, 2023-06-10:
Correction! It requires a refresh. I don't think it originally behaved like this but I  could be wrong.

nickcoutsos, 2023-06-10:
There's a websocket connection to the backend so as long as I'm receiving GitHub's webhook successfully it _should_ be pushing that info down to you.

It's been a while since I've tested the GitHub workflow end-to-end so I can have a look again. Has this been happening consistently for you?

WainingForests, 2023-06-10:
I haven't modified anything in a while, I pressed commit, got a drink came back and clicked it. Didn't realize I flashed with the old firmware and was utterly confused what happened for like 2 minutes. Using the refresh code button on the site didn't fix it, instead F5 solved it. It's very possible chrome or something on my end is at play. I just haven't changed anything afaik.

nickcoutsos, 2023-06-10:
> I haven't modified anything in a while

By any chance, had you left the browser tab open since then? The first possibility that jumps to my mind is that if the websocket connection is lost it won't attempt to re-establish it. That could happen if you go offline or if the server restarts (which would've happened a few days ago during a deploy).

There's definitely some room for improvement here, I could have it detect websocket disconnections and periodically attempt to re-connect. I could also have it discard the old build link when it sees that the latest commit doesn't match. But before all of that it would be nice to rule out this possibility.

> Using the refresh code button on the site didn't fix it, instead F5 solved it.

That would also make sense. The button reloads the currently selected keymap but I don't remember if it explicitly requests the latest build.

WainingForests, 2023-06-11:
Appears to be working fine. Not sure why that night was so weird. The one tab it just didn't notice the new build for like 20 minutes but they appear to be working fine. Might just have been a github overall thing anyway!

closed 2023-06-11 by WainingForests