nickcoutsos/keymap-editor #113: The new binding dialogue sometimes adds a shift to the letters I am trying to bind.

MikiVanousek, 2023-06-13:


MikiVanousek, 2023-06-13:
https://github.com/nickcoutsos/keymap-editor/assets/33402194/02d1c12d-09a4-42d6-92cc-d0841b07f2fe

MikiVanousek, 2023-06-13:
This is not the first time this happened. Last time it was just `&kp T` being changed to `&kp LS(T)`

MikiVanousek, 2023-06-13:
I can now clearly see in the video the `LSHIFT` is cheched.

nickcoutsos, 2023-06-13:
Hi! It sounds like you're satisfied already but I just want to make sure we're both aligned on expected behavior.

I guess for a start I should note that I've recently added a feature to toggle the modifier selection from keyboard input. So if you have that keycode selection dialog open and press the <kbd>LShift</kbd> key on your keyboard it will add that modifier to the bind. To me it seems like a small improvement and I recognize it comes with its own problems (mainly the possibility of interfering with browser/OS shortcuts) but it can also cause some confusion.

In your video it looks like you've set up some homerow mods but LS+LA doesn't seem intentional. Do you recall editing the _hold_ parameter's binding for that key?

nickcoutsos, 2023-06-13:
Nice, thank you! In my use I've never bothered to search using shifted characters (the search should be case-insensitive but also matching symbols like `%` instead of `PRCNT` is a fairly recent addition) but it's definitely a reasonable use-case and perfectly explains the issue.

It should also be _relatively_ straightforward for me to expand on that by ignoring modifier presses when that modifier was actually applied to another key. Maybe the feature will ultimately not be that useful, but this report will help me shape it into something that adds convenience without unexpected quirks.

I'll keep this issue open to track the issue.

nickcoutsos, 2023-06-13:
Okay, I've updated the event handler to only toggle modifiers if they aren't being held while another key is pressed.

closed 2023-06-13 by nickcoutsos