nickcoutsos/keymap-editor #115: missing buttons for "commit changes" and "firmware build" in web version

Niftelbin, 2023-06-15:
Hi, 
I used the web version of the keymap-editor to change my keyboard layout. Now I wanted to finalise the process and realised that the last two buttons ("commit changes" and "firmware build") are missing from the website. I am not a programmer, I just have a fancy keyboard, so if there is something wrong on my end - I have no idea where to look for it. 

[image]

nickcoutsos, 2023-06-15:
Hi! The _Commit Changes_ button is now _Save Changes_ and relocated to the top bar next to the keyboard selectors. It's disabled if the keymap is unchanged and quickly comparing the screenshot to the current state of your keymap in https://github.com/Niftelbin/zmk-config-xiao/blob/6992daad7e13004b8cb802433a7d304dfcd540ea/config/6x5_manuform.keymap it looks to be the same.

The firmware button depends on GitHub Actions (the editor doesn't do any compilation itself) and it looks like your repository doesn't have actions enabled. You'll need to set that up in https://github.com/Niftelbin/zmk-config-xiao/settings/actions

Niftelbin, 2023-06-15:
Thank you for the reply. I enabled "allow all actions and reusable workflows" and in Workflow  permissions "Read and write permission" as well as "Allow Github Action to create and approve pull requests", but the firmware button is still not showing up. Do you have another idea what I should try?

nickcoutsos, 2023-06-15:
The action hasn't run, you'll need to trigger that by pushing a new commit to the branch. If you don't want to do that manually you can just change a single key, save changes, change it back, save again.

Niftelbin, 2023-06-15:
I actually do not know what "pushing a new commit" means, but I have now changed a few keys and saved the changes. The button is still not showing up.

nickcoutsos, 2023-06-15:
Sorry for the confusion -- although I wanted this app to simplify things I've always assumed that users would be familiar with these tools.

I see that you've made changes to your keymap but I don't see a build. When you enabled the actions did you click the "Save" button underneath that list of options?

Niftelbin, 2023-06-15:
I found the problem. The Actions Permission site in Settings was not the only place I had to enable something. I clicked on the "Action" tab and it said that I had workflows disabled. Once I clicked on the green button for enabling workflows everything worked. Thank you for helping me. This is all very new to me.

nickcoutsos, 2023-06-15:
I'm glad to hear its working! Sorry for the slog, ZMK is still quite technically involved. In the future when it's possible to update a keymap without recompiling that should take a lot of that responsibility out of the user's hands.

closed 2023-06-15 by nickcoutsos